betaflight/betaflight-tx-lua-scripts #165: Minor textual improvement

0crap, 2018-08-29:
Is it possible to change, in the Betaflight / Rates (3/3) page, the text Anti-Grav Thr to Anti-Grav Thr**e** ?
Thr is often referred to as Throttle.
Adding only an e makes it much more clear it's about Threshold.

Just a small suggestion. Thx!

0crap, 2018-08-29:
[image]

klutvott123, 2018-08-29:
Good suggestion. I would have changed it to "thld" which is commonly used in aviation according to https://www.allacronyms.com/threshold/abbreviated

0crap, 2018-08-29:
No need to play it by the aviation book. ;-)

Coming from rc helicopters, in my mind that is Throttle Hold.
Oh well.....

mikeller, 2018-08-30:
I like this one: https://www.allacronyms.com/THRSH/Threshold

Nine out of ten pilots will read this as 'trash'...

0crap, 2018-08-30:
trash..., yes indeed. :-)

**Thrsh** is perfect, lets give that a go.

ijustwannafly, 2018-09-20:
rates3.lua

{ t = "Anti-Grav Thrsh",  x = 15, y = 30, to = SMLSIZE },

You can add the edit now or grab the latest build from the jenkins website.

0crap, 2018-09-20:
Thx! Tested and works great.
Can you please make a Pull request so it will end up in the official code?

Much appreciated.

0crap, 2018-10-22:
No stale bot please.
Still hope this can be done.

0crap, 2019-01-09:
Merged, so can be closed.
Thx all. 👍

closed 2019-01-09 by 0crap