betaflight/betaflight-tx-lua-scripts #167: Support OpenTX 2.2.2

ljfh001, 2018-09-03:
Doesn't seem to work with OpenTX 2.2.2. 
Getting 
attempt to index field 'protocol'

bfasty, 2018-09-18:
> I can launch it through the sd card file explorer menu. However, if it's launched as a telemetry screen, I get:
> Telemetry protocol not supported!
> Once, it got into that state, launching through sd card file explorer menu doesn't work any more.
> 
> tx: x9d+
> rx: R9 Slim
> OpenTX 2.2.2
> lua: 1.0.2

I am having the same issue with the same hardware,

Peschi90, 2018-09-19:
i have x9e with OpenTX 2.2.2 and everything works perfectly. just for Info

0crap, 2018-09-20:
> 
> 
> i have x9e with OpenTX 2.2.2 and everything works perfectly. just for Info

Also tested on Taranis Q X7S (OpenTX 2.2.2) using the SD card file explorer and execute from there.
Works perfectly.

0crap, 2018-09-20:
> 
> 
> I can launch it through the sd card file explorer menu. However, if it's launched as a telemetry screen, I get:
> Telemetry protocol not supported!
> Once, it got into that state, launching through sd card file explorer menu doesn't work any more.
> 
> tx: x9d+
> rx: R9 Slim
> OpenTX 2.2.2
> lua: 1.0.2

I was able to replicate your issue.

I copied an existing good working model in my radio.
After selecting the copied model, I deleted all telemetry sensors.
When I now long press the Page button, to call the lua script, I get the same error message.

Did you bind the receiver to the radio yet and can you see Telemetry sensors on your radio?
After a successful bind you can try "Discover new sensors" to see if that fixes the problem.

Let us know how it goes!

closed 2018-10-27 by stale[bot]

dm13dv1b, 2018-11-07:
Sad to see it closed. Does not work for me either. 2.2.2+BF3.4 no data on telemetry screen

Peschi90, 2018-11-07:
Then something goes wrong at your telemetry Sensors. Do you rename one sensor? Are all sensor detected? Try to delete All and let the Taranis finde them New

paulo-serrao, 2018-11-22:
I was getting the same error, after upgrading to 2.2.2

I went to the screen page where you configure the page to run the script, deleted the entry, a recreated it. To me it solved. (had to do this for every model)