betaflight/betaflight-tx-lua-scripts #168: PID extension for INAV

Peschi90, 2018-09-20:
Hello, 
first i love this script!! I dont need all of the functions but PID tuning Rates and so on, are the best functions in this script for me.

I had only one Problem. I use INAV and there are a lot more PID´s for the Nav things.
Becouse of that i prepared the script a little bit for me. I hope that is no problem for you.

My hope is that you want to help the INAV community too and you can Implement my prepared file for the other INAV users. I only changed the file fore the X9 recivers and test it at my X9E.

My Idea is that the file comes in one of the next releases and the user can decide wich file he want to activate. I dont think that it need a big implementation with menu and so on.

[pids.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2402461/pids.zip)

[image]

What do you think about it?

regrads Peschi

mikeller, 2018-09-21:
Happy to support an iNav version of these scripts. But considering that RAM on the TX is an issue, I think the Betaflight and iNav versions should be separate. Users will still be able to install both of them in parallel, and use either / or for individual models.
As well as this, iNav and Betaflight have different release schedules, and lua script releases are often tied to firmware versions, so having them in separate repositories (potentially as forks) that can release separately will be an advantage.

Peschi90, 2018-09-21:
yes or like u say. 
i only use the pid page so i cant say if everything else work correctly with inav. What i can see is that i have values at every page. the filters rates work definetly correct.
I use this script in combination with Inav Lua Telemetry and have no ram problems at my x9E.

My idea was that you put the PIDS.Lua in the BF folder and rename it like PIDS.lua_INAV. Then the user has to decide which of the files he want to use and uncomment the right on. 
That is the quick and dirty solution but a seperate repo is of course much better but also more work :)

Peschi90, 2018-09-21:
Wait i failed.
Rates doesnt work With inav. it looks like there are more differences as i thought. :/

closed 2018-10-28 by stale[bot]