betaflight/betaflight-tx-lua-scripts #176: GyroLowpass-Filter and dtermLowpass-Filter

gotsila, 2018-10-13:
Hello,
can anybody make a screen for the new Filtersettings in BF3.5?

GyroLowpass 1 and dtermLowpass 1 are there, but i dont know how to set the seconds of Filters.

Change the Notch-Screen or add a second Screen like the PID 1/2 screens.

Thanks a lot

mikeller, 2018-10-14:
@gotsila: The filter settings have been added already, and we are in the final stages of putting together a release of the scripts to go with the 3.5 firmware.

In the mean time, you can test 
[betaflight-tx-lua-1.1.0-rc1.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2475985/betaflight-tx-lua-1.1.0-rc1.zip). ;-)

gotsila, 2018-10-14:
nice to hear it. of course i will test it and give something back to the community.
But the  ZIP is empty, only an empty OBJ Folder is in it.

mikeller, 2018-10-14:
@gotsila: Oops, my bad. Try this one: 
[betaflight-tx-lua-1.1.0-rc1.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2476319/betaflight-tx-lua-1.1.0-rc1.zip)

gotsila, 2018-10-14:
OK. i have update all Folders and have tested. the LPF Gyro and DTerm on the Filterspage can now be saved.
Any other changes i can't see. i mean i can't see the LPF 2 for Gyro and DTerm. Theres only one Filters page with LPF 1 for Gyro and DTerm and the three Notches for Gyro1/Gyro2/DTerm.

gotsila, 2018-11-16:
Hello, i have checked the latest LUA-Scripts. All Options in there work fine, but i can't find the second Filter page with the LPF 2 for Gyro and DTerm.
How can i read the values with LUA ? if i can read and write the values, then i will try to program it for the comunity. i have knowledge in Java, Pascal, Basic and much scripting language, but LUA is new for me.

   read              = 92, -- MSP_FILTER_CONFIG
    write             = 93, -- MSP_SET_FILTER_CONFIG
    eepromWrite       = true,
    reboot            = false,
    title             = "Filters",
    minBytes          = 18,
	outputBytes       = 18,

0crap, 2018-11-18:
[image]
A picture makes it more clear I think.
Gyro Lowpass 2 Cutoff Frequency and Dterm Lowpass 2 Cutoff Frequency are missing in Version 1.1.0

closed 2022-01-17 by klutvott123