betaflight/betaflight-tx-lua-scripts #177: script syntax error telemetry protocol not supported

OxygenGS, 2018-10-13:
Hello,

I did install OpenTX 2.2.3 beta, because of the new flex firmware i try out for my R9.
However, i did a fresh install and setup, then did copy the BF lua files over, now when i press Page, it says: script syntax error telemetry protocol not supported

Can someone help me out please?

mikeller, 2018-10-14:
@OxygenGS: Try re-running 'Discover new sensors'.

ijustwannafly, 2018-10-14:
Is the lua script really dependant on the sensors?

I have the lua setup for no less than 8 quads.  All of them using smartport.

Combination of xsr-m and r9m.  All rx's update with latest firmwares.  Otx on xlite is official 2.2.  r9m lite and r9m rx are on the same flex firmware version.

Every now and then I get telemetry protocol not supported.  I have to change models to get it to "reset".

Sometimes it won't actually load any values and sometimes it will never save anything.

I have never done a discover sensor step ever since I started with the first 3 page lua script.

This is the biggest thing I dislike about the lua script...it's spotty on its connection reliability.  To add to it, I can't change vtx to raceband anything.  It always fails.

mikeller, 2018-10-15:
@ijustwannafly:

> Is the lua script really dependant on the sensors?

Since the data is tunnelled through the RX / TX via a 'sensor': Yes it is.

ijustwannafly, 2018-10-15:
> Since the data is tunnelled through the RX / TX via a 'sensor': Yes it is.

Well shoot looks like I will have to do that.  Like I said, on the old 3 page version (way old) I never actually discovered sensors for the lua to read data so I never did it with this version.

ijustwannafly, 2018-10-16:
Did a discover sensor on every single quad.

Still getting occasional "telemetry not supported" or no values loading on the page.

I noticed the protocol.lua file is using Tmp1 as the statesensor and also referencing RSSI values.

It makes me wonder why there isn't a reserved sensor (between betaflight and smartport at least) for dedicated lua communication.  

If the communication/data is being tunneled through an existing sensor I would imagine the original data the sensor is sending/receiving needs to be parsed out/in for the lua data to succeed.

Is this a proper assumption?

ijustwannafly, 2018-10-16:
@mikeller 

Ok I got that.  So then how frequent is the connection status refreshed?  As said before, with or without discover sensors the lua will read data.  It's just that every now and then it stops reading data or insists the protocol is not supported.  

I use only smartport.  Yesterday when testing the discover sensor requirement I attempted to change pid values and I got save, then retry, then blank pages on pid.  Moving to other pages (vtx, filters etc.) would load the values but going back to pids would have dashes for values.  I literally have to change models and back OR power cycle the tx.

I read the otx documentation on lua and there was a note that said telemetry scripts are loaded/initialized when the model on the tx is selected.  If that's case I would think there is maybe a refresh issue happening since I always have my tx and model selected before connecting a battery.  Also, the reason for changing models or power cycle when protocol not supported error occurs is because it stays in that cycle.  It doesn't seem to re-attempt to check the protocol again.

mikeller, 2018-10-18:
@ijustwannafly:

> Ok I got that. So then how frequent is the connection status refreshed? 

However often the background task is run by OpenTx.

ijustwannafly, 2018-10-20:
At this point I modded the lua to work with my xlite how I want it.  Since I have changed things in a way that would break function on other radios I am just going to continue exploring how to fix this refresh and protocol issue for my xlite.

Thanks.

ijustwannafly, 2018-10-20:
Mis-communication.   I meant I changed the lua, to this point, to allow digit changes to respond to long press.  I did submit the info in a thread (I am not setup to do pull requests) but since it involved changing some lines in ui.lua it didn't get picked up so I figured it was because it would break functionality on the other radios.

I also edited some of the pages (or dropped some) and changed other minor cosmetic items. 

For the refresh, I meant that since I have modded the lua to fit my use it's not a problem to try to figure out the refresh issue.

mikeller, 2018-10-20:
@ijustwannafly: The not being picked up is probably more due to nobody really taking care of the betaflight-tx-lua-scripts project at the moment...

If you point out the post with the changes that did not get picked up (or any other changes that you think should be made), I am happy to test on Taranis, and if they work open a pull request.

ijustwannafly, 2018-10-20:
#145 

I made the changes to events.lua and ui.lua.

I am still learning lua as I go along but if you care to review it by all means.

I will still be trying to figure out the refresh part when time permits.

klutvott123, 2019-12-18:
This is probably the right place for this since it's still open and pinned. 
Here's some more info about "script syntax error telemetry protocol not supported". 
The BF lua script uses the opentx ``sportTelemetryPush/crossfireTelemetryPush`` to detect which protocol is being used. This fails if the output telemetry buffer in opentx is not available.
The most likely reason is that there are other scripts running which are also sending data using these functions. For example when starting the tools version of the BF script when the telemetry version is already running this will often happen.

pattyteh, 2021-11-13:
This problem also occurs with TBS MPM (Multiprotocol Module) and the TBS Mambo in Multi FrSkyX2 Mode. In Multi FrSky Mode it works  for example.

The Telemetry is working fine. It must be due to FrskyX2 mode. 

Can someone help me to fix it respectively fix it maybe in a new version of the lua scripts to support the combination Multiprotocol Module and TBS Mambo?

klutvott123, 2021-11-13:
This isn't something we can fix in the Betaflight lua scripts. If you get "telemetry protocol not supported" when using FrskyX2 it's an issue in the radio firmware. 
```sportTelemetryPush``` will return ```nil``` if the telemetry protocol isn't recognized as frsky smartport.
https://github.com/opentx/opentx/blob/625f2b2f8394fec4f9b29d9627cf7c92df81b380/radio/src/lua/api_general.cpp#L492-L495

This is how we do the detection in the lua scripts
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/bad5cb347df587095bc4b3c3fa466b75b5795969/src/SCRIPTS/BF/protocols.lua#L30-L36

The reason for it working with your QX7 and not with the TBS Mambo could be that the mambo uses FreedomTx while the QX7 likely has OpenTx

closed 2022-01-17 by klutvott123