betaflight/betaflight-tx-lua-scripts #179: The gps rescue tab has no value

Pascal-m, 2018-10-23:
Hello,
I am in Betaflight 3.5.2 with a GSP and i do not see any value in the "GPS rescue" tab
My setup : 
Radio : X9D+
FC : F405 Matek
Thk

mikeller, 2018-10-24:
@Pascal-m: What RX are you using, and what firmware version is the RX running?

Pascal-m, 2018-10-24:
I use  
R9 MM 915MHz 
FW Version  :  180816

mikeller, 2018-10-24:
Looks like the R9mm suffers from the same problem with bytestuffing that was already identified (and partially fixed) on r-XSR and XSR:

https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-431221268

I have notified the FrSky engineering department about this, but have not heard back from them yet.

Closing this as it is a duplicate of #151.

closed 2018-10-24 by mikeller

mikeller, 2018-10-24:
No problem. Keep watching #151, I'll post updates and test versions from of new firmware from FrSky there as soon as I get them.

mikeller, 2018-10-24:
Yeah, that is my point about this being a duplicate - and in general it's easier for other users that come here when searching for a solution to this problem to have the entire discussion concentrated in one issue, compared to them having to read through multiple threads.