betaflight/betaflight-tx-lua-scripts #180: vtx values get overwritten at power up

janbolt, 2018-10-27:
the default values (Band F, Channel 2, 25mW) are send to the vtx at each power up.
Once changed, I would like my values to be stored.

RipperDrone, 2018-11-10:
did you save them or just change?

bobbfwed, 2018-11-10:
> Once changed, I would like my values to be stored.
Are you changing them using the Lua script (or OSD)? Or by pressing the button on the VTX?

RipperDrone, 2018-11-10:
Lua. Just wanted to make sure you set the values, then long-press menu and 'save'. OTherwise you just leave the page and values are NOT stored. Each individual Lua page needs to be saved separately...

In some cases, I have found deleting the vtx channel, freq, ... settings via CLI helped to retain Lua script values...

janbolt, 2018-11-10:
I have changed the values via lua script and saved them.

closed 2018-12-18 by stale[bot]