betaflight/betaflight-tx-lua-scripts #182: Display Distance, Speed, Mode, Battery etc

henfri, 2018-11-11:
Hello,

as far as I understand, these scripts can influence the PID values and the VTX settings.
But I think they cannot display Information like Battery status, Flight Mode etc., right?

Would you consider adding this as a Feature?
I am thinking of something like this:
https://uploads.tapatalk-cdn.com/20170215/9fcc412dddc41fc54b59281c21790afe.jpg

Greetings,
Hendrik

mikeller, 2018-11-11:
I think it is preferable to keep configuration and live data display in separate applications. Or what benefit do you see in combining them into one script?

henfri, 2018-11-12:
Hello,

I assumed synergies, but you can judge better.
How do you currently display the telemetry, by the way?

Regards,
Hendrik

mikeller, 2018-11-13:
@henfri: There aren't that many synergies, as the configuration script uses MSP over SmartPort to get / set data, and the telemetry script will be using telemetry data (available on FrSkyHub and SmartPort, and for firmwares that are not Betaflight) to get read only telemetry data.

henfri, 2018-11-14:
ok, thanks.

Out of curiosity: How are you displaying the telemetry? 
Some of the data (e.g. Flight mode) is encoded so that it is machine, but not human readable...
Greetings,
Hendrik

mikeller, 2018-11-14:
@henfri: I just use the telemetry displays built into OpenTX - telemetry is only secondary to me, as I am not looking much at my TX while flying.

closed 2018-12-21 by stale[bot]