betaflight/betaflight-tx-lua-scripts #185: Telemetry discovered but screen only shows ---

skorokithakis, 2018-11-22:
I have discovered 13 sensors on my R-XSR (but not F000), and telemetry screen is showing blanks (`---`) for everything. Am I doing something wrong?

mikeller, 2018-11-22:
What version of the RX firmware are you using? Try updating to one of these: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151#issuecomment-416183224

skorokithakis, 2018-11-22:
Great question, I don't know, I didn't change it after it came from the factory. I'll try that and report back, thanks!

SimplyKot, 2018-11-30:
I have the same problem:
reciever: XSR-M firmware FPORT_NonEU_171114
I have discovered 12 sensors anf after commend:
`set pid_in_tlm=on`  I got 13 sensors.

vkolotov, 2018-11-30:
Is there any way to debug lua script? Any logs that I can read?

vkolotov, 2018-11-30:
If it matters, I do not use software serial, I've connected receiver telemetry pin to a TX pin on my FC.

I'm pretty sure one day I saw "F000" channel, and lua scrip was working fine. I've upgraded FW everywhere and then it stopped working. I do not know what exactly FW upgrade caused this...

mikeller, 2018-12-01:
@SimplyKot, @vkolotov: Try the latest beta firmware for your RX as posted in #151, and report back please.

SimplyKot, 2018-12-02:
@mikeller
Sorry, but i have only XSR-M reciever.

mikeller, 2018-12-02:
@SimplyKot: To the extent of my knowledge the XSR-M uses the same firmware as the XSR.

SimplyKot, 2018-12-02:
Ok. I will test the new firmware today.

ijustwannafly, 2018-12-02:
This is an issue that I have been seeing since the lua was redesigned to the version it is now.

I have had very little time to really learn AND deconstruct the lua process flow.

I currently use xsr-m and r9m receivers.  All are updated with the latest frsky offering.

Using the latest betaflight.

With the "big" lua version sometimes I get dashes, sometimes a reboot loads values, sometimes it loads right away and sometimes I have to wait upwards of a minute for it to load on its own.

With the "small" lua version (old 3 page) everything just loads and works.  Even on the latest bf3.5.x.

The differences that stood out to me is that there wasn't a "background.lua" in the old version that constantly did a time check from what I found.

Also, the old 3 page version would load the msp version when the main ui page was loaded versus having it run in the background and/or when a page is changed.

It seems to me that this is more of a communication issue between how the lua interacts with smartport and not necessarily the firmware itself though latest firmware on the rx ensures bug fixes and issues have been resolved.

Just my observations.  A new job position recently has taken alot of my free time to play with this so I have just been flying more and defaulting to bf osd.  Though, bf osd is still limited and slower :(

vkolotov, 2018-12-02:
Good observation @ijustwannafly, I'd like to test the old version of the script as well, what version it is (that works for you)?

SimplyKot, 2018-12-02:
@mikeller 
I replaced firmware on my XSR-M reciever  and got the same situation.

ijustwannafly, 2018-12-02:
@vkolotov 

It was v0.3 on the release page.

Let us know your results of testing.

SimplyKot, 2018-12-03:
@mikeller, @vkolotov
Now I use latest SmartPort firmware XSR_FCC_180322.
v0.3 and v1.1.0 show blank (---) values.
Betaflight / SPRACINGF3EVO (SPEV) 3.5.3 Nov 15 2018 / 00:51:40 (61d5abf00) MSP API: 1.40
XSR_FCC_180322
Opentx 2.2.2 (a5de173b)

SimplyKot, 2018-12-03:
I just flash FC to inav 2.0.1
And it's work!
Working tabs are:
PIDs (1/2)
Filt (1/2) and (2/2)
PWM

klutvott123, 2018-12-08:
@vkolotov The reason for it loading the values when you do a reboot might be that for a short time after startup, the FC might not have any other telemetry data ready to send yet. In this short time there would only be the MSP request from the lua script, and the response isn't slowed down because of the constant telemetry stream. 

I remember reading somewhere that F000 is hidden in the latest opentx versions, so i wouldn't worry about that. I don't see it either with 2.2.2. 

I notice that some of you have a lot of sensors. As a troubleshooting step, can you try disabling the accelerometer? Do ``set pid_in_tlm=off`` also, since it only adds another sensor which will slow things down further. 

@SimplyKot Good idea to test with inav too. Your findings makes sense to me. 
As an example, here's the data that is sent back to the lua script when you enter the PWM page:

INAV: https://github.com/iNavFlight/inav/blob/dbdd1656a84ee4f09ebc14b4c0b86a9476a48f46/src/main/fc/fc_msp.c#L1010

BETAFLIGHT: https://github.com/betaflight/betaflight/blob/5c594a5562c29b98bdbdee6781f4521ad0662092/src/main/interface/msp.c#L1245

Betaflight sends back twice as much data in this case. 
To further back this up, I'm using a custom lua page to adjust LED brightness. It requires 64 bytes of data(even more with headers), and it takes forever to load 😄

vkolotov, 2018-12-09:
Thank you @klutvott123. I've disabled accelerometer and `pid_in_tlm`. I flew two hours today constantly monitoring if lua scripts working. It was very stable today, all values loaded fine before/after I changed batteries on my quad. Looks like your thoughts regarding telemetry is valid. However, I must say it was a windy day today, so this might contributed to the todays stability as it might cooled down the RX. I'll keep monitoring and let you know.

PS. I'm on the most recent BF dev build.

klutvott123, 2018-12-14:
This might help with the issues you guys are experiencing: https://github.com/klutvott123/betaflight-tx-lua-scripts/tree/discard-duplicate-data-from-buffer. 
Compiled version: 
[bf-tx-lua-discard-duplicate-data-from-buffer.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/2680437/bf-tx-lua-discard-duplicate-data-from-buffer.zip)
If anyone decides to try it, please let us know how it goes. 

@ijustwannafly I tested v0.3, and it is a little more "snappy". The pages in v0.3 doesn't require as much data as some of the pages in the later versions. Also after loading a page for the first time, it is saved for later, so as you cycle through pages it will load the values directly from memory. In the later versions this functionality has been removed, and i think it was done in an effort to reduce memory use. If this was to be done for all pages in v1.1.0 we would probably run out of memory. Since all values are stored as floats, it would require 100s of KB to store them all.

ijustwannafly, 2018-12-14:
@klutvott123

Thanks for the info on the v0.3 process flow.  Makes sense, storing and loading values to/from memory.  With the size of the bigger lua it would definitely run out of memory earlier.

I had an idea, based on how the lua has been growing, to have a "lua builder" script that allows you to choose which pages to load out of the available options.

For example, with the new GPS features I don't need them on my proximity quads but I do need them on my LR quads.  Would be nice to have a file that allows me to choose which pages to use on that model and then create the page (x7pre.lua for example).  Again my new work position has taken time from me to play with this idea.

klutvott123, 2018-12-14:
@ijustwannafly I like your idea. Personally i just comment out all the pages i don't need, but i understand that might be tedious for some. I imagine something that let's you choose between different profiles, where each profile has "slots" where you can just put in whatever you want, pids, rates, etc. Then you could  quickly switch between predefined setups or tie a setup to a model.

ijustwannafly, 2018-12-14:
@klutvott123

I do the same thing where I edit the x7pre.lua file but sometimes in the field I find I commented out a page I thought I won't use only to need it then.

mikeller, 2018-12-15:
@ijustwannafly:

> I had an idea, based on how the lua has been growing, to have a "lua builder" script that allows you to choose which pages to load out of the available options.

Doing so to only show pages to users that they are interested in makes sense. But since every page is unloaded from memory after the user has moved to the next page, this will not actually do anything to improve the situation with respect to memory usage.

mikeller, 2018-12-15:
@ijustwannafly: As I said, makes sense.

vkolotov, 2018-12-28:
Ok, just a heads up. I've been using official and "fixed" version of FW for R-XSR. The results are not consistent, sometimes it works whole day sometimes it starts acting up right away.

I've built another quad, with absolutely the same components but different frame which has more room  so I separated rx from other components and it never gets "---" now, very stable. 

Which makes me think that the issue in temperature. E.g. when rx is close to VTX it is acting up, when it is separated, it works great.

Pounii25, 2018-12-30:
Hi all, same problem here.I got lua script page on the Taranis but not the data from my quad. I got telemetry fine. Here my set up:

-FC running BF 3.5

-R9 MM with the latest stable firmare for F.port ( 180629 ) FCC

-R9m-lite with the latest stable firmare ( 180625 ) FCC

-FrSky X-Lite with OpenTx 2.2.2 stable release

-Lua Script 1.1.0

Every help is appreciated, thanks :)

klutvott123, 2019-01-03:
@vkolotov The issue might very well be that these receivers are sensitive to temperature, and that some are affected worse than others. As i described in the other thread, if the receiver sends a frame and doesn't get a acknowledge response back from the transmitter, it will retransmit the frame. This doesn't matter much for regular telemetry, but it causes the reception of MSP data in the lua script to fail even if all the data is received by opentx. The retransmits shows up as duplicates that cause a sequence number mismatch. It might be possible that this happens more frequently at higher temperatures. Did you try the version of the script i included in a post further up? 

@Pounii25 You might also want to try the script included here: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/185#issuecomment-447350543. Not guaranteed to fix it, but worth a try.

Sybalau, 2019-01-14:
Same problem with R-XSR, beta 3.5 and script 1.1.

Only VTX page work, the other show ---.

pljeroen, 2019-02-10:
Using an R-XSR myself here. I think people are getting farked on F4's without the ability to invert. This is not LUA related, but a simple fact of life with an F4. Set up SoftSerial, done.
I have been going crazy on my Racerstar STARF4S to get it to properly work using UART6. I finally just hooked smart audio to that UART, which does work normally, and created a SoftSerial on the LED_STRIP (had to free some resources) Then just use that SoftSerial for SmartPort and go on with your life.

pljeroen, 2019-02-10:
I DO have another issue. When I power up the radio first (X-Lite) and then the quad, I sometimes have to reboot the radio, or else my RSSI will remain 0 on my OSD. In that scenario, my PID values are --- 
Reboot the radio, and everything works like it should, but I dont like the idea of a powered on quad with a powered off radio. An option to reload the script would be great.

mikeller, 2019-02-10:
@pljeroen: Reloading scripts is something that has to be solved in OpenTX, not in the script itself. Please open an issue for this in https://github.com/opentx/opentx/issues.

klutvott123, 2019-02-27:
Just found out that MSP over telemetry has been disabled for some targets in BF 3.5 to make the firmware fit in flash. For example SPRACINGF3EVO has MSP over telemetry disabled. This means that lua won't work.

mikeller, 2019-02-27:
Very good find, @klutvott123, I had not even considered that. But yes, the flash space shortage has come to a point where even basic features like MSP over SmartPort has had to be removed from some F3 targets to make the firmware fit.

klutvott123, 2019-02-27:
@mikeller :thumb: 
It's understandable. I guess features directly related to flight and equipment support has to be prioritized to make this work for as many setups as possible.

mikeller, 2019-02-27:
@klutvott123: It's always going to be judgement calls with respect to what is removed and what not, we don't have much data on what is actually used. What we want is for users to spend a couple of dollars more and move to an F4 or F7, where we don't have to make those decisions.

klutvott123, 2019-02-27:
@mikeller That would be ideal. I suspect the only way to make it happen would be for manufacturers to stop making F3 boards. Easiest way for that to happen is for betaflight to just pull the plug on F3s. In some cases the F3s are actually more expensive too. One of the stores i use sells the BETAFLIGHTF3 for $1 more than the F4 version. 😆

mikeller, 2019-02-28:
@klutvott123: See https://github.com/betaflight/betaflight#end-of-active-development-for-stm32f3-based-flight-controllers.

klutvott123, 2019-02-28:
@mikeller I saw the PR just after my above comment. Immediately thumbed up 👍

Sybalau, 2019-03-02:
So no way to enable MSP over telemetry on F3 with betaflight 3.5?

mikeller, 2019-03-02:
@Sybalau: It is enabled for most targets, and only disabled on targets where there is no space for it.
You'd have to do a custom build of the target you want, disabling the things you don't want, and enabling the ones you want.

Sybalau, 2019-03-04:
@mikeller : I have no idea how to do this. Do you have a link to a tutorial?

mikeller, 2019-03-04:
https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

closed 2019-04-10 by stale[bot]

skorokithakis, 2019-04-23:
Does anyone here know which `define` we need to use to get MSP over SmarPort? I enabled `USE_MSP_OVER_TELEMETRY` but nothing happened.

klutvott123, 2019-04-23:
@skorokithakis ''USE_MSP_OVER_TELEMETRY'' is what has to be defined to make the lua script work. Defining it in ''common_pre.h'' for F3 mcus or in ''target.h'' for your target should work. Which BF version are you making the changes to? Which FC do you have? BF 3.5 and 4.0 work a little different when it comes to removing features.

skorokithakis, 2019-04-23:
I have the SPRACINGF3, I'm using BetaFlight 4. Do I need to enable telemetry output on my UART in the configurator as well?

klutvott123, 2019-04-23:
If you're using SBUS/smartport I think you need to have the uart configured for smartport and the telemetry feature enabled. With FPORT you get telemetry "for free".

skorokithakis, 2019-04-23:
I see, thanks. Unfortunately, I think I don't have a free UART, since I use GPS. I'll look at the controller again, thanks!

skorokithakis, 2019-04-23:
Oh wow, turns out my receiver supports FPort, so I should be able to use that for telemetry, thanks @klutvott123!

klutvott123, 2019-04-23:
@skorokithakis Yes. Then you can have it all on one wire. Betaflight wiki has good info on how to set it up 👍

pljeroen, 2019-04-24:
Oscar Liang as well. All my telemetry issues went away after switching to f.port. Highly recommend.

skorokithakis, 2019-04-24:
Fport worked perfectly, thanks everyone!

BradB111, 2019-05-18:
This is happening for me on crossfire nano diversity Rx with latest firmware,

klutvott123, 2019-05-21:
@BradB111 Which FC do you have?

BradB111, 2019-05-21:
> @BradB111 Which FC do you have?

Funny you should ask. I'm using the MATEK F722-STD. Due to you asking I have just tested on another quad i have with the same FC but uses an R-XSR, also doesn't work. 

FC is common on both failures.

klutvott123, 2019-05-21:
@BradB111 I was just thinking maybe you had one of the F3 boards that has msp over telemetry disabled. It should work with your setup. I also have a Matek F722 std with r-xsr using FPORT running BF 4.0,0 and it works. I haven't tried the latest BF version yet. Did it work before?

BradB111, 2019-05-21:
> @BradB111 I was just thinking maybe you had one of the F3 boards that has msp over telemetry disabled. It should work with your setup. I also have a Matek F722 std with r-xsr using FPORT running BF 4.0,0 and it works. I haven't tried the latest BF version yet. Did it work before?

exact same setup here. I cant recall it working previously, it may have done I cant remember unfortunately. that quad is also on BF 4.0.0

klutvott123, 2019-05-22:
@BradB111 Just to narrow it down I would start by reflashing the FC and do a full chip erase, configure the serial port and FPORT only.

BradB111, 2019-05-22:
> @BradB111 Just to narrow it down I would start by reflashing the FC and do a full chip erase, configure the serial port and FPORT only.

Ok reflashed quad, full chip erase as always, configured port, selected fport, Set SerialRX inversion on, set serial half duplex on, rx now works in receiver tab.

Results:
Telemetry working fine
BF Lua not working

BradB111, 2019-05-25:
@mikeller can this be reopened?

klutvott123, 2019-06-03:
@BradB111 Then I think we can assume everything is right at the FC side. You could try deleting all sensors from your transmitter and rediscover them. If that doesn't work then I don't know what else to try.

BradB111, 2019-06-04:
> @BradB111 Then I think we can assume everything is right at the FC side. You could try deleting all sensors from your transmitter and rediscover them. If that doesn't work then I don't know what else to try.
That didn't work so I tried the following:
Clean flash Quad (full erase) to BF 4.0.3, configure serial port and set FPORT under configuration to get Tx+Rx talking to FC. Install latest Lua version on Tx, deleted all sensors. power on Tx and Quad, discover sensors -> all sensors displayed fine. Go into lua. Still nothing

[12 more replies]