betaflight/betaflight-tx-lua-scripts #197: Syntax error on OpenTX 2.2.3

ljfh001, 2019-01-15:
I keep getting Syntax error with OpenTX 2.2.3. Does the current version work with OpenTX 2.2.3?

BavariaAirview, 2019-01-18:
Hi, 

Installed OpenTX 2.2.3 on my Taranis x9d+.
Downloaded the Lua 1.1.0 and modified the pre9x lua for vtx page to be first.
Then I deleted the pre9x.luac and copied everything to the taranis. 
Works fine here. 

Best Regards

ljfh001, 2019-01-18:
I tested with R9 + R9 mini. It does work the first time when I launch it, but after launching it again getting syntax error complaining about radio not defined.

ljfh001, 2019-01-18:
I can see the lauc files got modified after the first run, then the next time when I launch it stopped working.

klutvott123, 2019-01-21:
Sounds like opentx compiles the script even if the luac files supplied with BF lua script are up to date. Did you have ``luac`` checked when you downloaded opentx? If so, have you tried with ``luac`` unchecked?

ljfh001, 2019-01-21:
Yes, I have luac checked for opentx. I also tried with luac unchecked. It worked the first time when I launched it. The subsequent runs causes out of memory errors.

klutvott123, 2019-01-21:
Strange. ``radio not defined`` would indicate that ``radios.lua`` wasn't loaded. Can't see any obvious reasons to why, but something might have changed in version 2.2.3. There are other reports of similar problems with 2.2.3.

ljfh001, 2019-01-29:
I finally got more time do some additional testing. I think this is related to the old 'out of memory' error. I'm using a x9d+. Everything now works if I just delete the horus and x7 dir. The LUA scripts have grew the memory footprint by a lot with the addition of all these new features. I think we're at the limit again.

klutvott123, 2019-01-29:
But when it runs out of memory, it will just stop the script and tell you that it ran out of memory. I just tried 2.2.3 on my QX7 + R-XSR with no issue. Did it work with 2.2.2? Tried it without flex firmware? 
Problems with R9 + flex firmware + 2.2.3 in #177 too.

ljfh001, 2019-01-29:
I did have r9flex checked for opentx 2.2.3. Maybe that uses additional memory. 
From my observation, it seems opentx 2.2.3 recompiles lauc's at runtime even though the pre-compiled lauc files already exist. I think what happens is that the luac files got corrupted when it runs out of memory. I see sometimes the bad lauc files are 0 in size.

joriws, 2019-03-05:
Was the error, because I get this:
Script syntax error
TELEMETRY/bf.lua:13
<eof> expected

Because I get this error. I did x9pre.lua change as in posting #2 vtx to be first..

I've 2.2.3 with lua and luac..

closed 2019-04-11 by stale[bot]