betaflight/betaflight-tx-lua-scripts #199: requiredVersion does not match actual API version

hsuq510, 2019-03-12:
in the gpspids.lua the string requiredVersion = 1.041 does not match the API version, therefore the gpspids is not displayed.

hsuq510, 2019-03-12:
it seems apiVersion is always 0

hsuq510, 2019-03-12:
as background is run before ui, we suggested to fix and run the check inside the main task like:

function run_ui(event)

   if not dataInitialised then
         if data_init == nil then
              data_init = assert(loadScript(SCRIPT_HOME .. "/data_init.lua"))()
         end
     dataInitialised = data_init.init();
     if dataInitialised then
           data_init = nil
           collectgarbage()
     end
   end

mikeller, 2019-03-16:
@hsuq510: Have you tested this? This will result in the UI not showing and the script locking up whenever there is no connection, which I suspect most users will find problematic.

laz2wiringpi, 2019-03-17:
i have no gps screen with build 27 and the rel  has the screen but does not work with 4.0.
IMO
 Should it even be enabled ?
 is it  coded for the changes in 4.0 ? there were many ... i might be wrong though.

laz2wiringpi, 2019-03-17:
Fyi 
revert  the code for requiredVersion for a test . 
 gps screens appear. the gps values  seem to work and the values seem to align . 
I used x7  -  requiredVersion  works on other transmitters ?

hsuq510, 2019-03-17:
> @hsuq510: Have you tested this? This will result in the UI not showing and the script locking up whenever there is no connection, which I suspect most users will find problematic.

Yes I have tested with no connection and the UI is showing , no problems in my tests

laz2wiringpi, 2019-03-17:
@hsuq510 
can you make a pull request and or a ref to your test branch ?

klutvott123, 2019-03-18:
The code works for me as it is now. BF 4.0.0 RC2, QX7 with latest lua master. The only way I can reproduce this issue is to turn on the transmitter, navigate to the UI and then power on the FC. In that case ``run_bg`` hasn't received the api version yet and the gps pages will not be visible. Since ``run_bg`` isn't allowed to run while the UI is visible, the api version won't be received until you navigate out of the UI, let ``run_bg`` do it's thing and navigate back into the UI. Then the gps pages should be there. 

I can imagine it's a little annoying/confusing. Another way to deal with it could be to allow ``run_bg`` to run until the api version is received even if the UI is visible.

hsuq510, 2019-03-18:
My test fail with opentx 2.2.2 and X9D+, as the run_ui get before run_bg I thought it was a race condition issue.

laz2wiringpi, 2019-03-18:
No luck 
https://ci.betaflight.tech/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/27/artifact/

turn on  transmitter,
plug in FC BF 4.0.0 RC2 omnibussF4 

waited 10 seconds just in case
then got to bf lua script 2 press of page as mine is second page 
got data for first pages ...
press center button no  gps pages . 
no idea how it worked for @klutvott123 
firmware on the Q7 - https://www.open-tx.org/2018/07/28/opentx-2.2.2

navigated out of ui 
waited 60 60 seconds 
still no gps pages .. 

strange  ..... no idea ...

klutvott123, 2019-03-18:
@laz2wiringpi That is strange, yes. I just tried it with the lua version you linked to and it still works for me. It should only take a few seconds for it to receive ``apiVersion``. I'm using opentx 2.2.3 by the way. Forgot to mention.

RRAlex, 2019-03-31:
Just wondering, but in a mixed BF 3.x / 4.0RC drone setup, should the lua (master) from the CI be used to support 4.x or can 1.1.0 already support both versions? (X9D+)

laz2wiringpi, 2019-04-15:
i think this can be closed 
works crosfire Make sure you are using CROSSFIRE  2.94 BETA if you have nano diversity receiver
and r9 900mhz
 and  2.5mhz.  

with latest scripts

closed 2019-05-22 by stale[bot]