betaflight/betaflight-tx-lua-scripts #208: Rssi drops from OSD (without rxloss)

pljeroen, 2019-04-12:
**Describe the bug**
During mid-flight and sometimes after battrery swap without turning transmitter off, my OSD's RSSI value drops to 0 and starts flashing. A transmitter reboot is required to regain RSSI values in OSD. Other telemetry values are unaffected.
It happens at 1:35: https://www.youtube.com/watch?v=NOYSGuwPBi4

**To Reproduce**
Random behaviour. Will fly 3 packs straight without problems, and then all of a sudden manifest again. I even loose the value mid-flight, without RXLOSS and I can just continue flying. 

**Expected behavior**
I expect this to NOT hapen, as there is no RXLOSS.

**FlightBF4.0 now, redone the complete config from scratch:
```# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr 11 2019 / 07:01:44 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name
name PLJ Rev3

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 27 3 900 1300 0 0
aux 4 13 5 1700 2100 0 0
aux 5 28 0 1700 2100 0 0
aux 6 35 2 1700 2100 0 0
aux 7 36 3 1700 2100 0 0
aux 8 39 4 1925 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -17,38,38
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_switch_mode = KILL
set bat_capacity = 1500
set yaw_motors_reversed = ON
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_vbat_pos = 14409
set osd_rssi_pos = 14425
set osd_remaining_time_estimate_pos = 14380
set osd_flymode_pos = 14402
set osd_g_force_pos = 14765
set osd_throttle_pos = 14658
set osd_vtx_channel_pos = 14786
set osd_current_pos = 14415
set osd_mah_drawn_pos = 14807
set osd_craft_name_pos = 14795
set osd_stick_overlay_left_pos = 14661
set osd_stick_overlay_right_pos = 14674

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
 - CL RACING F4S v1.5 Betaflight 4 RC4, but happened on 3.5.6 as well.
 - r-xsr receiver with smartport hooked up as per CL RACING manual. FCC180921.frk
 - X-Lite transmitter with LUA script installed, OpenTX2.3.3

**Additional context**
It happens at 1:35: https://www.youtube.com/watch?v=NOYSGuwPBi4

[image]
[image]
[image]

I'm new to reporting bugs or issues here, so please let me know this needs more info, and in that case, what. At your service for debugging purposes.

mikeller, 2019-04-14:
@pljeroen: I would almost suspect this is caused by a bad connection of the RSSI wire between your RX and the flight controller. Please check your wiring. RSSI is not used for anything like RXLOSS, and has only informational character.

pljeroen, 2019-04-14:
There is no RSSI wire. RSSI is being fed back somehow through the transmitter. When the transmitter is off, no RSSI value at all. If I power on the transmitter after the quad, I will get RSSI reliably.
Sometimes it just drops from OSD. 

Using s.port telemetry.

mikeller, 2019-04-14:
@pljeroen: Ok, this is using RSSI over MSP through the lua script then. And it looks like the lua script stops sending updates (or they are lost somewhere in between). Can you make sure that you are running the latest firmware on your r-XSR (from https://www.frsky-rc.com/r-xsr/). Or better yet, switch to FPort, which will give you in-band RSSI support.

pljeroen, 2019-04-14:
@mikeller just did that, initial tests look promising. I was already on the latest version for rx firmware, but switched to the fport version of it. Tnx for your time!

pljeroen, 2019-04-21:
Issue gone when using f.port. Thanks for the suggestion guys!

closed 2019-04-21 by pljeroen