betaflight/betaflight-tx-lua-scripts #209: NV14 Target not supported

discord5000, 2019-04-13:
Support for the Flysky Nirvana would be great! xfire.lua is now working in the flysky firmware...

mikeller, 2019-04-13:
@discord5000: Good suggestion. Do you have a link to the OpenTX source for the NV14's firmware to make it possible to integrate with it?

discord5000, 2019-04-13:
https://github.com/chevyhu/Flysky-OpenTX-Test-Branch

mikeller, 2019-04-14:
There are no releases on this since 2017. I do not think that it makes much sense to put effort into supporting firmware that is almost 2 years old, I think this should be closed.

discord5000, 2019-04-14:
?? Last commit was 2 days ago. Everyone is running bleeding edge on this radio.

mikeller, 2019-04-14:
So where are the releases for this 'bleeding edge' then? Developing scripts for the small percentage of users of this radio who are set up and willing to build OpenTX every few days sounds like a waste of time.

discord5000, 2019-04-14:
facebook and rcgroups. the devs are fairly active on rcgroups, too. https://www.rcgroups.com/forums/showthread.php?3023832-UndergroundFPV-Dark-Knight-Radio

mikeller, 2019-04-14:
I still can't see anything coming even close to organised releases. Trying to support something that runs on what is essentially a constantly moving target is mostly futile and a waste of time. We can look into this once they have managed to sort out releases / been merged back into OpenTX.

closed 2022-01-17 by klutvott123