betaflight/betaflight-tx-lua-scripts #215: GPS Rescue layout mix-up

Jetrell, 2019-04-26:
Presently -
`Ground speed` is displayed under `Throttle Minimum` on the script.
`Hover Throttle`  is displayed under `Ground speed` on the script.
`Throttle Minimum `is displayed under `Hover throttle` on the script.

[image]
[image]

Are these figures just in the wrong location and GPS rescue can still be tuned from the script?

klutvott123, 2019-04-26:
@Jetrell You're right. Those parameters were mixed up for the X9. They can still be tuned from the script. Just have to be aware that they are mixed up.  Fixed in #216. Try it out if you want and thanks for letting us know! 👍

Jetrell, 2019-04-26:
Thank you for the fix.  I'll give it a go once its merged to the master.

PR #216 (2019-04-28): fix gps rescue for X9
Throttle min/hover and ground speed were mixed up for X9. Fixes #215
closed 2019-04-28 by mikeller