betaflight/betaflight-tx-lua-scripts #219: X-Lite Pro Support

SilentZen, 2019-05-10:
X-Lite Pro needs
["xlites"] = supportedPlatforms.x7,
in radios.lua

3djc, 2019-05-11:
Tested ok with that change on Xlite Pro and upcoming OpenTx 2.3

mikeller, 2019-05-11:
@3djc: Can you open a pull request please?

3djc, 2019-05-11:
I'll do, I need to check what platfrom X9 Lite is returning, so that we kill two birds with one stone :)

3djc, 2019-06-10:
Closed by pr #220

anishnattamai, 2019-06-11:
How do I fix it on the xlite pro? it is not working?

mikeller, 2019-06-11:
@anishnattamai: Please post your issues only once - posting them multiple times will not increase the likelihood of others being willing to spend their time to help you.

3djc, 2019-06-12:
It is working fine here on xlite pro on OpenTX nightlies using ACCESS or genuine D16 receivers. Multi and cloned receivers are giving issues, but probably neither betagflight or OpenTX can fix this, only FrSky

viktorglekler, 2019-06-22:
on OpenTX 2.3.0N30 the joystick up and down does not select the value to set. If I want to change PIDs I can change only ROLL P because it is highlighted and selected first but I can not select the next value I or D.. And if I press enter, I can change the value only downwards by moving the joystick up. On earlier releases of OpenTX it worked fine. The same for all other settings like VTX

mikeller, 2019-06-23:
@viktorglekler: Please report bugs in OpenTX nightly builds at https://github.com/opentx/opentx/issues/new/choose and not here.

3djc, 2019-06-23:
@mikeller We have reviewed all radio key handling in 2.3 because it was not very consistent, so 2.3 onwards radio report only their physical keys, and also a set a virtual keys to help script dev to do multi radio script. Maybe we can work together on fine tuning the virtuals if they need further improvements

mikeller, 2019-06-23:
@3djc: Makes sense - we normally focus on supporting the latest released version, so we'll rework the key bindings in the scripts for the first release after OpenTX 2.3 has been released.
Do you have a link to the specification of the virtual keys in OpenTX?

3djc, 2019-06-23:
Doing a quick browse, it would look something like this. Where there might be an issue is plus/minus break, since it is a granularity that rot enc cannot give

```
return 
{
    press = {
        minus = EVT_VIRTUAL_PREVIOUS,
        plus = EVT_VIRTUAL_NEXT,
        pageDown = EVT_VIRTUAL_NEXT_PAGE,
        pageUp = EVT_VIRTUAL_PREVIOUS_PAGE
    },
    longPress = {
        enter = EVT_VIRTUAL_ENTER_LONG,
        menu = EVT_VIRTUAL_MENU_LONG
    },
    repeatPress = {
        minus = EVT_VIRTUAL_PREVIOUS_REPT,
        plus = EVT_VIRTUAL_NEXT_REPT
    },
    release = {
        enter = EVT_VIRTUAL_ENTER,
        exit = EVT_VIRTUAL_EXIT,
        menu = EVT_VIRTUAL_MENU,
        minus = ?,
        plus = ?
    },
    dial = {
        left = EVT_ROT_LEFT,
        right = EVT_ROT_RIGHT
    }
}
```

closed 2019-07-30 by stale[bot]