betaflight/betaflight-tx-lua-scripts #221: Crossfire Nano Diversity - Blank values, working Telemetry

BradB111, 2019-05-19:
Hello,

Recently I have built a long range quad using the Crossfire nano diversity receiver. 

I am using:
-1.2.1 version of the Lua scripts
-2.2.3 OpenTX on a Horus X10S
-Latest FW for Crossfire Nano Diversity RX
-Latest FW for Crossfire Micro TX

The Problem: 
Using the latest Betaflight LUA Scripts whenever I enter into the Lua script (standard and compiled versions .lua/.luac) none of the values are updated all showing '---' on all of the pages. However I do not see the 'no telemetry' warning at this point and all telemetry values outside of the LUA script are working as intended such as GPS / Voltages / attitude / mAh Consumption / Current etc...

This also only seems to happen on my Nano Diversity RX, my crossfire nano receivers (non diversity) and R-XSRs (all using FPort) run fine. 

Any help would be greatly appreciated and if there is any more info that I can provide to help please let me know.

I honestly have no ideas on what could be causing this at the moment besides compatibility issues.

mikeller, 2019-05-19:
Duplicate of #198.

@BradB111: Please refrain from cross-posting the same issue multiple times - this is just making people less inclined to help you as you are wasting their time.

closed 2019-05-19 by mikeller

BradB111, 2019-05-19:
Well one would assume a stale issue from 3 months ago with no feedback and as a result has been closed would very unlikely be revisited hence why I posted a new issue. 

How am I to know if it's been abandoned or if it has been acknowledged and a fix in in the works?

With no replies, no further information needed and a closed topic it's like this has been abandoned and nobody was intending to take the matter further.

hiepdinhxxx, 2019-06-25:
My RX nano show the BF lua script blank values too :(

hiepdinhxxx, 2019-06-25:
@BradB111 Do you fixed this issue?

BradB111, 2019-06-25:
> @BradB111 Do you fixed this issue?

you need to update to 2.95 for The nano diversity rx (currently only beta though so I don't use it yet) but 2.93 works for the non-diversity nano rxs.