betaflight/betaflight-tx-lua-scripts #223: X9 Lite support

czuryk, 2019-06-01:
Need adding X9 Lite support.

mikeller, 2019-06-01:
@czuryk: Can you please open a pull request for this change? Thanks.

klutvott123, 2019-06-01:
It was added in #220
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/1417c399941983bb15ffb02858db620a6f066dd7/src/SCRIPTS/BF/radios.lua#L35

mikeller, 2019-06-02:
Sorry @klutvott123, I've kind of lost track in the flurry of hardware that FrSky is releasing. :stuck_out_tongue_closed_eyes:

klutvott123, 2019-06-02:
@mikeller So many s, lites and pros lol

mikeller, 2019-06-02:
@czuryk: Looks like it's already supported - can you close this please?

czuryk, 2019-06-03:
> 
> 
> @czuryk: Looks like it's already supported - can you close this please?

I am sorry but I still dont see release with with issue....
https://github.com/betaflight/betaflight-tx-lua-scripts/releases

mikeller, 2019-06-03:
@czuryk: There you go: https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/1.3.0

czuryk, 2019-06-03:
Excellent!
Thanks!

closed 2019-06-03 by czuryk