betaflight/betaflight-tx-lua-scripts #224: When connected to open tx I get  "Firmware update check failed" new version info not found or invalid

itsmerandy, 2019-06-04:
Hello, Purchased a new Taranis X-Lite and went through process of upgrading the firmware and had issues with OpenTx giving me error messages everytime I open it with controller connected. nIt doesn't matter if controller is in Bootloader mode or turned off or on I still get this message.
"Firmware update check failed, New version information not found or invalid."  
Also I get an SDcard error on x-lite when I turn it on  "Expected version: 2.2V0016"  The Sdcard has a newer SDCard version 2.3V0019 on it, why is it expecting an older SDcard ver?
Sorry, I'm really new to all of this. Thanks

mikeller, 2019-06-04:
@itsmerandy: This is the Betaflight project's issue tracker - we do not do anything with the firmware on your TX. https://github.com/opentx/opentx is the right place to open an issue for your problem.

itsmerandy, 2019-06-04:
Sorry.  Can I just copy and paste it over there?

From: Michael Keller 
Sent: Monday, June 3, 2019 9:39 PM
To: betaflight/betaflight-tx-lua-scripts 
Cc: itsmerandy ; Mention 
Subject: Re: [betaflight/betaflight-tx-lua-scripts] When connected to open tx I get "Firmware update check failed" new version info not found or invalid (#224)

@itsmerandy: This is the Betaflight project's issue tracker - we do not do anything with the firmware on your TX. https://github.com/opentx/opentx is the right place to open an issue for your problem.

mikeller, 2019-06-04:
@itsmerandy: Knock yourself out.

closed 2019-07-11 by stale[bot]