betaflight/betaflight-tx-lua-scripts #225: Does not work with xlite pro

anishnattamai, 2019-06-11:
I can put it on the xlite pro sd card but when I turn on the controller and power on the quad, I dopnt get any information about pids and vtx channels, I only see some lines and I cant change anything. is there a way to fix this, is there a new version comeing soon?

mikeller, 2019-06-12:
@anishnattamai: First of all, describe your configuration, what flight controller, firmware version, firmware configuration, RX, RX firmware, OpenTX version are you using.

Secondly, does your telemetry work? If not then there is no point in even trying to get the lua scripts to work.

anishnattamai, 2019-06-13:
@mikeller Im using a Rxsr on access and a xlite pro running the latest stable version, a lux f7 fc on bf 4.0.3, my telemetry is working I think because I clicked discover new sensors and I got 14 sensors. Thanks in advance!

mikeller, 2019-06-15:
@anishnattamai: Have you tried using the script again after you did the 'discover sensors'?

closed 2019-07-22 by stale[bot]

Hawxxer, 2019-08-06:
Not working for me too. I use Crossfire Micro + Nano Receiver. 
What I tested:

Quad with BetaFPV 12a AIO +  Betaflight 3.5.3 + Crossfire Nano
-> On X-lite Pro the Script says no Telemetry, But I find Telemetry Sensors
-> Using the same Quad with a FrSky X9D Plus and the Same Crossfire Module the Script is working fine.

Other Quad with Kiss Setup + Crossfire
-> Kiss Lua Script is working fine on the X-lite Pro

But I am not sure if it is a Bug of the Nightly Opentx or the this Skript

anishnattamai, 2019-08-06:
After I updated my xlite pro, it started working

Hawxxer, 2019-08-06:
Not working just updateded to the newest nightly, but I saw that there is an open Opentx-Issue for this problem.
https://github.com/opentx/opentx/issues/6580