betaflight/betaflight-tx-lua-scripts #227: Menu button on Taranis X7 does not work running OpenTX 2.3

vkolotov, 2019-06-16:
I'm testing OpenTX 2.3 and lua scripts (latest release - 1.3.0). I can switch to a display with lua script and I can see my PIDs, however I cannot switch to other screens or save changes. Effectively "menu" button does not work, when I push it (either long or short) nothing happens.

Is there any way to debug this issue?

mikeller, 2019-06-16:
As far as I can tell there is no 'OpenTX 2.3' release - is it possible that this is just a bug in a development build? You should probably report this as an issue against OpenTX.

klutvott123, 2019-06-16:
https://github.com/opentx/opentx/pull/6502 maybe. Looks like they're in the process of implementing "virtual keys" for LUA. Probably still work in progress.

vkolotov, 2019-06-16:
Thanks guys, I'll close it.

closed 2019-06-16 by vkolotov

vkolotov, 2019-06-17:
Just for a reference: https://github.com/opentx/opentx/issues/6509