betaflight/betaflight-tx-lua-scripts #228: some pages of the script have not data

mainline, 2019-06-16:
some pages of the script have not data 

for example : there are two pid tuning page at the very beginning of the script,  page 1 of 1 has data but 2 of 2 has no data

any ideas?

klutvott123, 2019-06-16:
Which receiver? Which firmware version is on it?

mainline, 2019-06-17:
hi klu thanks for helping me ive pulled chunks of hair out and have spent an embarrassing amount of time trying to figure this out...

frsky rxsr - R-XSR_FCC_190311

mainline, 2019-06-17:
Configurator: 10.5.1
Firmware: BTFL 3.5.7
Target: FURYF4OSD
Open TX 2.2.3
LUA script have tried 130, 121 and 120

mainline, 2019-06-17:
on the 2 of 2 pid page in the script all of the values are --- 

if you try to enter in a value this error occurs:

Script syntax error
?:0
attemp tp perform arithmetic on field 'val

klutvott123, 2019-06-17:
@mainline The problem is that BF 3.5.x doesn't have some of the new options added to the script. D Min for example. You can upgrade to BF 4.0.x or try version 1.1.0 of the script

mainline, 2019-06-28:
Thanks Klu im sorry for the late reply.  Thank you! upgrading solved all my problems

closed 2019-08-04 by stale[bot]