betaflight/betaflight-tx-lua-scripts #232: OpenTX 2.3 tools shortcut support

3djc, 2019-06-28:
Opentx 2.3 will introduce a new way of accessing key LUA, in the form of a tool menu:

[image]

I think BF script would be a perfect match for that feature. To get in the list, a script must have a lua file in /SCRIPTS/TOOLS/ that idealy include a line for giving it a friendly name in that form : 
local toolName = "TNS|My lovely BF script|TNE".

That file can be the script itself (one way of achieving it could be to move the whole directories into TOOLS dir), or simply a laucher like you do with /BF/bf.lua,

mikeller, 2019-06-28:
@3djc: Sounds interesting. Will this finally make it possible to run scripts in the background (like telemetry scripts do) and have them react to user input on the X10 / X12?

Do you want to open a pull request to add this, or should I have a go at it (I am not really set up to test though).

3djc, 2019-06-28:
Mike, sorry no, this does not change that. We have a complete redesign of colors screens handling planned for 2.4, part of that will adress your question, but unfornatly not before I'm afraid.

I did not raise a PR because there are a couple of different ways this can be approached, and I'm not sure which one you prefer.

As for testing, it should not be too bad to test, you can run 2.3 simulators by installer 2.3 companion nightly, that should be enough to test the launching mechanism.

Please be aware that this lauching is not yet available on Horus, I had planned to do it this morning, but other urgencies came my way. This week end hopefully !

mikeller, 2019-06-29:
Thanks @3djc, we'll go with the flow here and add things to the scripts as they become supported in OpenTX.
I think for this change, adding a launcher like `/BF/bf.lua` into `/SCRIPTS/TOOLS/` is the right thing to do.
We can potentially consider moving the scripts into a better location in a separate pull request, but I am not sure that this is a good move - it will not improve much for users, and it will potentially cause confusion if users just add a new version of the scripts to their SD card, and leave the old scripts in place, making diagnosing problems harder.

closed 2019-08-05 by stale[bot]

mikeller, 2019-10-27:
@3djc: Follow up question to the addition of the 'Tools' page: Is there a plan to make it possible to start a particular tool with a 'shortcut', or assign it to a button / switch?
While the tools menu in itself is very useful, it takes quite a bit of navigation to get there, so users might find it useful if they can define 'shortcuts' to open the tools they use most often.

tubbers, 2019-11-04:
Agree with mikeller. Without a shortcut to the Tools menu, there is no advantage over the telemetry screen approach to access (actually less convenient)