betaflight/betaflight-tx-lua-scripts #234: multiple syntax errors

gitseco, 2019-07-14:
I've tried to install different versions of this script on my x9d with varying results, but I'll focus on my issues with the 1.3 release. I've had several different syntax errors with this release alone. 

First, when I installed it and used the Page shortcut to enter the script, I successfully got to the PIDs page, the next few pages were fine, then on the 5th page, I get "Syntax Error: Attempt to index upvalue '?' (a nil value)

Then I tried to load the script from SD Card Tab in the Radio Setup Menus. I couldn't even open that one, it says "Telemetry Protocol not supported." Which is odd, since I do have telemetry available. I tried to delete the sensors and then discover new sensors, and the same 13 sensors show up and I still get the same issue. Also, when I open through the Page shortcut, as mentioned above, I am able to see my current PIDs, etc, it's not blank. 

Then I uploaded an older version of the script, and that had it's own issues with memory and other stuff, and still didn't work, Ieven tried the luac script method some people mentioned. So then I reinstalled 1.3, and then I got a whole new message on like the 7th or 8th page, where it said something about not being able to find 'pwm.lua', but I can see that it is on the sd card where it should be. I would give the exact message, except it only happened a few times and I didn't write it down, and I don't seem to be able to reproduce it, now I always get the index upvalue.

The main reason I want this script is to adjust my vtx channels and bands. Except I'm unable to even get to that page because I get blocked before I get there. Does anyone have any thoughts or suggestions on how to fix this?

mikeller, 2019-07-14:
@gitseco:
- what version of Betaflight are you using? Make sure it is the latest (4.0.4);
- from your description, it looks like something went wrong when installing multiple different versions on top of each other - wipe and reinstall your SD card, and install the latest version of the script on it.

gitseco, 2019-07-14:
Thank you for the response.

I am on betaflight 4.04

I also have my taranis on the latest 2.2.4

I will wipe and reinstall the sd card to see if that helps. But I was having this issue before I did multiple installs, it was happening right away. I only tried older installs to see if it would help. Do you have a process or recommendation for wiping and reinstalling? I've just been bringing the SD into my computer and deleting all the files, uploading all the taranis opentx 2.2.4 files, then adding the lua script to it. Is that good, or is there a cleaner way to wipe it?

rachoban, 2019-08-08:
Not sure if anyone is still looking at the post. Check out Joshua Bardwells latest videos on LUA scripts, there is a way to made your VTX LUA script come up first when you press the page key

closed 2019-09-14 by stale[bot]