betaflight/betaflight-tx-lua-scripts #235: Problem with LUA scripts not showing values on all LUA scripts

rachoban, 2019-08-08:
I'm trying to figure out why I can only see **data** on some of my LUA scripts, 
The scripts that don't show data are below and the other LUA scripts show values:

Betaflight /PIDs (2/2) values don't show,
Betaflight / Rates  values don't show, 
Betaflight /PID Advance  values don't show, 
Betaflight / Filters  values don't show

I have 3 quadcopter with this problem 
1. **Chameleon-1**    JB AIO Flight controller (latest version), betaflight 3.3.3, R-XSR receiver (R-XSR firmware: R-XSR_FCC_190311.frk)
2. **Chameleon-2**    JB AIO Flight controller (latest version), betaflight 3.5.3, R-XSR receiver  (R-XSR firmware: R-XSR_FCC_190311.frk)
3. **Chameleon-TI**    Kakute7, betaflight 3.5.5, R-XSR receiver  (R-XSR firmware: R-XSR_FCC_190311.frk)

Taranis Configuration
Running OpenTX 2.2.4 on my Taranis X9D +
SDCard is: sdcard-taranis-x9-2.2V0021
LUA script is: betaflight-tx-lua-scripts_1.3.0
Firmware is: opentx-x9d+-noheli-lua-luac-en

Betaflight: Configurator: 10.5.1

Question 1: Do you have, or know of someone who has a quadcopter that is setup with any of the above Flight controllers and the R-XSR receiver, that can  see all LUA scripts **data** when viewing the LUA scripts.
Question 2: Do you have, or know of someone who has a Taranis X9D + that is setup with the above configuration who can see all LUA scripts **data** when viewing the LUA scripts.
Question 3: Do you have any suggestion where to go from here to solve my problem

Thanks for your help

klutvott123, 2019-08-08:
The solution is to upgrade to BF 4.0.x or try version 1.1.0 of the script. The pages that are not working for you have been updated with parameters that are only available in BF 4.0 and above.

rachoban, 2019-08-08:
Klutvott123
Thanks for the response, Do you have all the LUA scripts working on a quadcopter with the my taranis configuration and using BF 4.0.x. Is it alright to use sbus on the R-XSR or should I go to Fport

klutvott123, 2019-08-08:
@rachoban Yes, I do. Only difference is  opentx 2.2.3 on Taranis QX7 but it shouldn't matter. Sbus should be perfectly fine. BF lua script also works a little faster with sbus.

closed 2019-09-14 by stale[bot]