betaflight/betaflight-tx-lua-scripts #243: VTX values

raso12, 2019-09-16:
BF 4.1.0 the values of the vtx are not adjustable

Raimund

robschii, 2019-10-01:
Need vtx table set in betaflight

patryk-jason, 2019-10-01:
Not exactly. I've set up table correctly. Channels can be changed via LUA, power levels not.

mikeller, 2019-10-02:
@raso12: Can you please supply information on what VTX model you are using, your CLI `diff` from Betaflight, and what version of the lua scripts you are using on what TX model / firmware?

frizzle101101, 2019-10-05:
ive got same issue on x9d+ legacy, bf4.1RC5, using atx micro 3, using script version 1.3

mikeller, 2019-10-05:
@frizzle101101: Try upgrading OpenTX to 2.2.4 (or better) - there was a bug with telemetry in lua in 2.2.3: https://www.open-tx.org/2019/07/11/opentx-2.2.4

kaweksl, 2019-10-10:
same issue. I can't change any vtx settings, rest is working fine. I used vtx table for IRC Tramp EU from BF wiki.
Does the current version suppose to work with vtxtables at all ?

X9D+ legacy with opentx 2.3.1, Betaflight / KAKUTEF4V2 (KTV2) 4.1.0, bdir dshot600

kaweksl, 2019-10-10:
After some trying i found out that in BF settings you must set Band, Channel and Power to something, if not you want be able to change anything in VTX thru lua. 
Also script expect specific table, probably that hardcoded in BF 4.0.6 ? .

raphaelcoeffic, 2019-10-14:
@mikeller what is actually the status here? I don't quite understand. Is it a bug, a feature request, something that needs more doc?

mikeller, 2019-10-15:
@raphaelcoeffic: I would see this as a feature request, but we need to figure out what we want to do. With the introduction of VTX tables the old way of having hard coded VTX frequencies / power levels in the released version of the lua scripts isn't viable any more. Getting the tables dynamically from the flight controller probably isn't either, given the memory size and connection speed constraints.
Maybe we could add a facility to the Betaflight configurator that allows the user to export their custom VTX table as a .lua file, and then drop it into the script folder on their TX?
Thoughts?

raphaelcoeffic, 2019-10-15:
Well, if the tables are available / configured in the configuration, then it makes great sense to have them generated there. I’ll try to have something working in LUA. If you could give me some pointers, I could give it a try in the configuration as well.

mikeller, 2019-10-15:
@raphaelcoeffic:

Have a look at https://github.com/betaflight/betaflight-configurator/blob/master/src/js/tabs/vtx.js#L515-L567. This is the function that saves the VTX table as JSON, and most of it can be reused. We'd probably have to write our own function to go in place of https://github.com/betaflight/betaflight-configurator/blob/d7e65b86e17cd4c9284baefc5acb214481032f10/src/js/tabs/vtx.js#L546-L547 and assembles a lua table instead, but that shouldn't be too hard. And then we can have `vtx.lua` check for such a file, and read the table from there.

raphaelcoeffic, 2019-10-15:
@frozenskys not at all! It's most probably the right thing to do! And having these tables generated by the configurator is the perfect fit. So the question is only how to link it to the model name? Or maybe the user will have to rename it by hand, or something like that.

raphaelcoeffic, 2019-10-15:
sure, but the configurator does not have the model name ;-) Anyway, having a fallback is a good idea, I like it.

frozenskys, 2019-10-15:
no we don't have the model name, but we do have the *craft name* and if they happen to match then happy days :)

mikeller, 2019-10-15:
Good going, guys! I had not even thought of the issue of having to have the same set of scripts support multiple different models.
We _could_ use MSP to get the craft name from the flight controller, to get a reliable match to the craft name in configurator - but not sure if we aren't over engineering then.

rickcote0041, 2020-01-18:
Hi guys good discussion, I have the immutable vtx values via lua script issue too. I'm running betaflight v4.12 and running opentx v2.2.4 on a QX7s attempting  to exec bf lua script v1.3... is there a solution to try yet or release candidate I should test?

frizzle101101, 2020-01-18:
Update opentx yo 2.3.4 and ur reciever firmware ...everything is working
fine on the latest firmwares

frizzle101101, 2020-01-18:
If its just the power values that are wrong labeled then you need to edit
the lua script to change the vtx power labels duno if theres something for
that yet

rickcote0041, 2020-01-23:
Thanks not sure what the root cause is because I have it working for 2
other quadcopters with crossfire receivers and bf 4.1.0 working. It may be
that I selected the wrong smartaudio version for the not working when I
downloaded the vtx table. I'll recheck the version using the blackbox test.

Max-Headroom, 2020-01-30:
OK, I'm confused. In the Readme it says:

> If you are using a VTX that supports the SmartAudio or Tramp protocols then bands and channels etc. are managed using vtx tables since firmware version 4.1.0. The betaflight configurator will allow you to save your configuration to a craftname.lua file. 

Where does BF do that? Is this correct? Or should I manually edit the vtx.lua file and the save it as craftname.lua

frizzle101101, 2020-01-30:
That was the original idea idk if anyone ever implemented it

klutvott123, 2020-01-30:
Support for VTX tables will be in the next release of the script/configurator. If you want it now you have to use the master branch and the latest nightly build of the configurator. You generate ```craftname.lua``` in the configurator. Then you copy it to ```BF/VTX``` on your SD card. The script will then automatically use the tables from that file.

sgiesRC, 2020-02-12:
Hi,

I am new here and I got a question which could maybe answered by the people here which knows the most about the BF Lua issues I maybe have and ma be it is a bug or something...

I updated to BF 4.1 and after the update I have a problem with smartaudio. I have a picture in my goggles, so thats working.

I use the TBS Unify Pro (V3 I think). I measured the cable from TX to the UART 1 and the connection is okay (and worked before on the old BF 3.5 and on the same quad). I just updated to BF 4.1.3 and I have the newest OpenTX version on my X9D+.

I followed Joshua Bardwells tutorial, but I do not get the VTx Settings working via Lua Script on my Taranis.

Via Sensors page and debug I got the value 217 (where Joshua B. gets 216) and this should be "SmartAudio 2.0 unlocked".

Here is his tutorial I followed step by step.
https://youtu.be/VRQJJTyHNJw?t=466

I set the vtx tables like he did via downloading a *.json file for SmartAudio 2.0 (I tried 2.1 as well just to be sure thats not the issue) and I tried the manual version with copy paste the CLI command which I get from Oscar Liangs page.

The json file gives me a band called "UNKNOWN" with 8 frequencies all saying "0". The other one  I got doing it like Oscar Liang directly over CLI does not have this band. Both do not work in my telemtry screen and do not work if I start the BF lua manually and not over my telemtry page on my taranis.

As well i went to telemetry page in the Taranis and chose again "Script" >> "bf". Again not working.

What could be the issue? Is there a newer version of the BF lua or something where the older BF lua does not work with VTX tables?
Or is it just working WHEN the quad is armed?

If this post here is not the correct way or place to ask please just say it. I am new here.

[image]
[image]
[image]
[image]
[image]

Max-Headroom, 2020-02-12:
While you only talk about the VTX, you don't clearly say if your smart audio is working. Can you do other settings, like PID?
And can you set the settings in your goggles using your TX sticks?
See if you can see and set your VTX in goggles.
See the reply to my post before yours about lua.

sgiesRC, 2020-02-12:
I can see my settings in my telemtry screen where I look at BF lua. The one with the goggles  I did not try yet.
[image]

sgiesRC, 2020-02-12:
--UPDATE---

I tried to enter via OSD Menu and there I could change the VTx settings. Wasn#t easy to tell the OSD that I want to enter. The OSD menu was visible a second and disappeared after that. After a few trials I managed it to get into the menu.

Maybe my Sticks are not centered correctly (I fly Mode 1)???

After I entered OSD menu I changed VTx settings there. Saved and exit.
And suddenly I can change VTx settings in the Taranis via LUA-script (which wasn't possible before; see post bottom of this post).

BUT: it is easy to change the channels UP. Like Channel 4 to channel 5. But back from channel 5 to channel 4 is very hard. I have click very fast.

Seems like not a BF bug or something, but I have no idea why is it like that

Max-Headroom, 2020-02-12:
> After I entered OSD menu I changed VTx settings there. Saved and exit.
> And suddenly I can change VTx settings in the Taranis via LUA-script

Don't trust this yet. Wait until they do the upgrade. In my case, it looks like you can change it through lua. I found it was not the case. My VTX went into fail safe mode(25W). I've also dumped that VTX since, also for other reasons.
Can't help you with the sticks, works for me in mode 2, and I'll be using that till the next release of configurator and lua script comes out.

sgiesRC, 2020-02-12:
Thx for your fast answer. Which upgrade you are talking about?

Max-Headroom, 2020-02-12:
> Support for VTX tables will be in the next release of the script/configurator. If you want it now you have to use the master branch and the latest nightly build of the configurator. You generate [...]

For your sticks problem, it is not related to this issue, best you go ask on the forums (Oscar Liang's Into FPV comes to mind).

dmyers7, 2020-02-18:
Running BetaFlight 4.1.2
Configurator 10.7 to export the VTX tables as Craft.Lua (As per the name set the Configuration Tab) (VTX Table downloaded from OEM and working)
Running OpenTx 2.3.5 on a Jumper T16 Pro
Using CrossFire and Telemetry is working fine
Downloaded and install BF script 1.4.1
Install the Craft.lua (Larva_X.Lua in the case) into the root \BF\VTX Folder and then added it to \SCRIPTS\BF\VTX when the first directory appeared not to work.  Tried Larva_X.LUA and Larva X.LUA

BF Script continues to use the default settings which if you choose a bad option locks up the script or the VTX from making further changes until it is reset in Configurator; and of course the Power Settings do not reflect the real values (plus has na extra for this VTX).

Help is appreciated.

mikeller, 2020-02-18:
@dmyers7: Support for custom VTX tables (#264)\has not yet been released. If you want to use this you'll have to build it from the repository.

dmyers7, 2020-02-20:
@mikeller That was a bit of a summersault, but impressive.  I now have VTX tables working on my T-16 under the tool menu with the compiled BF.lua Master branch and the as yet to be released BetaFlight Configurator.  Very nice table interface as a "tool" so you don't have to page through all the menus.  Confirmed working with the VTX tables on my Larva which has 3 power output levels now all named correctly and it shows it is changing with an LED color.  Thank you all for the hard work and the code refactoring.  Will play with the rest as I get time.

mikeller, 2020-02-20:
@dmyers7: I hear you - our discussion has got me to think about continuos integration to this project to get to a point where an installable development version is built after every change (we already do this for Betaflight configurator and blackbox explorer, both of which are much more complex).

klutvott123, 2020-02-20:
@mikeller That would be nice and would make it much easier for people to test if they want. We do have the jenkins builds, but it's using the old ```build.sh``` and is currently producing builds without the needed files. Is that something that has to be updated manually?

mikeller, 2020-02-21:
@klutvott123: What is missing? The Makefile is using `build.sh` as well, and that one is complete.

klutvott123, 2020-02-21:
https://ci.betaflight.tech/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/83/artifact/obj/SCRIPTS/BF/
```/SCRIPTS/BF/COMPILE/scripts.lua``` is missing. It has also created .luac files which the new ```build.sh``` doesn't do.

dmyers7, 2020-02-21:
@klutvott123 Your saying the Jenkins builds are incomplete.  If you're able to, pull the master branch from this GitHub and build it; it has all the correct files you need and builds a proper obj folder.  Placing these files on your radio, on first launch of the BF app from the Tool menu in OpenTx it will compile all the lauc files on the radio at once and launch; again no errors.  (You do have to make sure the lauc option is checked for your radio OpenTx firmware build as I recall).

klutvott123, 2020-02-21:
@dmyers7 Yes, Jenkins is using an older build script for some reason. Building it from source is a solution, but I think Jenkins should do a better job! 😄

Jansen330, 2020-05-05:
> @mikeller That was a bit of a summersault, but impressive. I now have VTX tables working on my T-16 under the tool menu with the compiled BF.lua Master branch and the as yet to be released [...]

How did you get that!! i saved the craft.lua, putted on the SCRIPT/BF/Q7! Thats my radio and continues the same in the lua script in my radio!

klutvott123, 2020-05-07:
@Jansen330 That's the wrong folder. The new version of the script supports vtx tables and will download them automatically. It will be released very soon.

Jansen330, 2020-05-07:
> @Jansen330 That's the wrong folder. The new version of the script supports vtx tables and will download them automatically. It will be released very soon. 

Ah ok!! That’s awesome!! Tks a lot to advise

bonoboxita, 2020-06-27:
Hi i'm having the same problem.
I'm using a RushTank Mini, F722 Flywoo controller, lastest lua, latest opentx and BF 4.1.7
I can read the VTX value, i have the correct VTX table on the SD card in the radio but i can't change any values.
All the other function i can enter and change.

mikeller, 2020-06-28:
@bonoboxita: Update to 4.2 and the latest Betaflight TX lua scripts.

jettero777, 2021-01-07:
Hello,
I'm using BF 4.2.5, Taranis X Q7 with last OpenTx 2.3.10, last lua scripts 1.5.0
After script initialization VTX table has downloaded well. I can select different parameters but they doesn't change in the FC.
If I change power or channel, theb exit and open script again, the changed parameter reverts to old, also in OSD I can see that it didn't changed.

klutvott123, 2021-01-07:
@jettero777 You have to save the changes you make. Change the values you want to change. Then long press enter to bring up the menu. Select "save page"

jettero777, 2021-01-07:
@klutvott123 tnx, now it works :)
strange, but there is no mention about it in the script doc
https://github.com/betaflight/betaflight-tx-lua-scripts

klutvott123, 2021-01-07:
@jettero777 Nice 👍 
You're right. Maybe we should add a "How to use" section to the readme. As it is now there's no way for the user to know the menu is there unless they experiment or watch youtube videos.

mikeller, 2021-01-09:
I think this can be closed as this has been implemented.

closed 2021-01-09 by mikeller