betaflight/betaflight-tx-lua-scripts #244: VTX Page - Lose Power and Pit after gaining telemetry

JonPetitta, 2019-09-23:
Kakute F7 (BF 4.0.6)
Atlatl V2 (?)
Tekko32 F3 (32.6)
R9MM - FPORT (20190201)
R9M (20190117)
X7S (2.2.4)
tx-lua (1.3.0)

I just installed and used the scripts for the first time today.  All was working when I first started.  I had screen 1 set to script bf.  I was having some issues with RSSI on OSD, so added num (also tried bar) to screen 2 to display RSSI.  When long pressed Page, bf script would start, but also would flash once with RSSI on bottom (think this was from bar mode).  I then removed script from screen 1, and was able to see RSSI.  I resolved OSD issue by setting scale.  Then set screen 2 to none, and added bf back on 1.

Now, only on VTX page (all others are fine) Power and Pit does not display on screen while receiving telemetry.  Though if I scroll past Band/Chan, it appears to go to power/pit though I can not see anything on that section of the screen.  It will then go to Dev if I keep scrolling.

If I load script while quad is powered off, all of the items on the vtx page display.  Once I plug in quad, Power and Pit disappear.  If I power off, and reload from script, they reappear.  Power and Pit did show, prior to me messing with num/bar on screen 2.

I've tried re-flashing OpenTX and re-copying the scripts, which did not help the issue.  I also tried copying the model to a new slot, didn't help either.

mikeller, 2019-09-23:
@JonPetitta: Does controlling your VTX through the OSD work?

Can you please post the output of a `diff` (enclosed in 3 backticks (\`\`\`)) in the Betaflight CLI here?

JonPetitta, 2019-09-23:
I am unable to get into VTX TR in the OSD menu.  I think I was able to get into there previously.  This is a new build, and I'm in the process of getting it all setup.

Here is the diff
```
diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.6 Sep  1 2019 / 00:44:18 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 E09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature TELEMETRY

# beeper
beeper -BAT_LOW

# beacon

# map

# serial
serial 6 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 28 2 1700 2100 0 0
aux 4 35 3 1300 2100 0 0
aux 5 39 2 900 1300 0 0
aux 6 47 1 1700 2100 0 0

# adjrange
adjrange 0 0 4 900 2100 29 4 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -60,-15,29
set min_check = 1010
set max_check = 1990
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT1200
set align_board_roll = 180
set ibata_scale = 275
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set pid_process_denom = 1
set osd_rssi_pos = 6562
set osd_tim_1_pos = 6582
set osd_flymode_pos = 6616
set osd_ah_sbar_pos = 4302
set osd_ah_pos = 4174
set osd_current_pos = 6603
set osd_warnings_pos = 14377
set osd_avg_cell_voltage_pos = 6572
set osd_core_temp_pos = 6594
set osd_stick_overlay_left_pos = 8518
set osd_stick_overlay_right_pos = 8527
set osd_profile = 2

# profile
profile 0

set feedforward_transition = 40
set f_pitch = 120
set f_roll = 120

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 40
set roll_rc_rate = 126
set pitch_rc_rate = 120
set yaw_rc_rate = 200
set roll_expo = 20
set pitch_expo = 40
set roll_srate = 65
set pitch_srate = 50
set yaw_srate = 60

# end the command batch
batch end

# 
```

mikeller, 2019-09-24:
No surprise there - you have not configured your flight controller to control a VTX:

[image]

GitHub is for discussion around the _development_ of Betaflight. For questions around setup and configuration, please use other online resources like RCGroups.