betaflight/betaflight-tx-lua-scripts #245: opentx 2.3 number jumping back to another value

robschii, 2019-09-27:
for example you want to switch vtx channel.......  number is jumping to the value before    or you hold the plus/minus button for a long time

hsuq510, 2019-09-29:
sames here, it seems the EVT_MINUS_BREAK is not properly handled. Any value changed with the minus button on a X9D+ (not 2019) has the effect when release the minus button the value returns the previous without any effect.

step to reproduce:
select a value, change it with minus , release minus button , see value change back to what is was at first

robschii, 2019-10-03:
nothing happend till now

RipperDrone, 2019-10-03:
https://github.com/opentx/opentx/issues/6852#issuecomment-538082073

RipperDrone, 2019-10-03:
searched in ui.lua (not sure this is the right spot...) - looks ok so far doesn't it?
[image]

RipperDrone, 2019-10-03:
incValue() - is this part good?

[image]

hsuq510, 2019-10-03:
betaflight event.lua has the following:

    dial = {
        left = EVT_ROT_LEFT or EVT_UP_BREAK,
        right = EVT_ROT_RIGHT or EVT_DOWN_BREAK
    }

removing the OR condition make script to work properly. This is not an opentx issue.

RipperDrone, 2019-10-03:
maybe the Horus or Taranis 2019 wheel commands are mixed up with the old button commands the wrong way...?

hsuq510, 2019-10-03:
can you please test if this in the events.lua fix the issue ?

dial = {
    left = EVT_ROT_LEFT ,
    right = EVT_ROT_RIGHT
}

RipperDrone, 2019-10-03:
No, it doesn't :-(

mikeller, 2019-10-03:
This code was added in #135 by @raphaelcoeffic. Maybe he has got an idea what needs to be done here?
As it is now with these scripts, we are supporting a number of different hardware types with different switches / toggles / wheels, so we need to thread carefully and do thorough testing to make sure we don't just shift the breakage to different hardware.

RipperDrone, 2019-10-03:
Yes, it does (I first kept the wrong half of the expression, thinking the ROT_ identifiers were supposed to fit the Taranis 2019 with ROTation wheel... :-))

So dropping the ROT_ parts of the OR expression does the trick on my Taranis 2013 X9D+

hsuq510, 2019-10-03:
Yes, to me is working.

RipperDrone, 2019-10-03:
Maybe the Taranis 2019 identifiers are pre-initialized to values which make the OR expression go wrong for a Taranis 2013?

hsuq510, 2019-10-03:
well it seems like opentx 2.3 has wrongly assigned EVT_DOWN_BREAK assigned to minus

raphaelcoeffic, 2019-10-03:
> As it is now with these scripts, we are supporting a number of different hardware types with different switches / toggles / wheels, so we need to thread carefully and do thorough testing to make sure we don't just shift the breakage to different hardware.

We have discussed this internally with the other opentx devs and it seems that some events have been broken for LUA since 2.3. We will need to look into it.

However, it should be time to think about moving to the new virtual events, which should normally bring compatibility with all radios with minimum effort.

mikeller, 2019-10-03:
@raphaelcoeffic: You seem to be in the loop on OpenTX development, would you be able to propose a change to move to virtual events?

raphaelcoeffic, 2019-10-03:
Yes, That’s the plan. I don’t have much hardware to test on, but I guess a lot of people are eager to test ;-)

mikeller, 2019-10-03:
@raphaelcoeffic: Well, the hope would is that if we move to virtual events that are then mapped to hardware switches / wheels by OpenTX then the mapping to the different hardware types would have been done in OpenTX when they were added. :wink:

raphaelcoeffic, 2019-10-03:
@mikeller that’s the idea, yes. The feature is already in OpenTx, so we need to adapt the key mapping, and that’s it.

3djc, 2019-10-04:
I just reviewed the virtual events to make a difference between INC/DEC for changing value, and PREV/NEXT for navigating the UI, since x9d(+) has those reversed. Should be merged shortly

mikeller, 2019-10-04:
@3djc: I like that. Can you alert us / propose a change when the fix for X9D+ has been merged - I think a considerable part of our users are using this hardware.

RipperDrone, 2019-10-04:
@mikeller @raphaelcoeffic Here it is: https://github.com/opentx/opentx/pull/6890 (thank you @3djc !) :-)

raphaelcoeffic, 2019-10-06:
first PoC for the virtual events: #250

mikeller, 2019-10-08:
@raphaelcoeffic, @3djc: Any chance that this is fixed in 2.3.1?
If so I think the right thing to do is to release a version of the lua scripts that include the changes needed for Betaflight 4.1 now, and tell users to update to OpenTX 2.3.1.
This will then allow us to give the 'virtual events' changes in #250 some weeks time for thorough testing / feedback, and after that we can do another release that includes them.
Thoughts?

raphaelcoeffic, 2019-10-09:
@mikeller Just added #251 for this purpose, let's see if that works better. It does for me on the simulator, but I don't have any X9D+ for testing in real...

closed 2019-10-10 by mikeller

RipperDrone, 2019-10-10:
no, not resolved yet on X9D+ ... https://github.com/betaflight/betaflight-tx-lua-scripts/issues/255