betaflight/betaflight-tx-lua-scripts #247: Lua Script Taranis X9D+ Plus 2019

wascheck, 2019-10-04:
I received this radio today and tried to install the lua script v1.3.0  and it did not worked.
The message :
Script Syntax error
Radio not supported
X9D+2019

raphaelcoeffic, 2019-10-04:
The code has already been added, but not yet included in a release. Sorry for that, but you’ll have to wait a bit.

wascheck, 2019-10-05:
Ok....thanks a lot for the answer. Gona be waiting the new version with it
fixed.

AndyJack0764, 2019-10-05:
Hi, similar issue here with X9D+2019,   would the older version work meantime?
tks

wascheck, 2019-10-05:
Works perfectly with my QX7 with Opentx V2.3.....

Hillgoonfpv, 2019-10-12:
Is there an ETA as to the release for this?

raphaelcoeffic, 2019-10-12:
@Hillgoonfpv whenever there is a next release, which should happen in short.

raphaelcoeffic, 2019-10-14:
@Hillgoonfpv 1.4.0 provides support for the X9D+ 2019. Please re-test directly with the ZIP file in #260, as it contains some fixes.

closed 2019-10-14 by raphaelcoeffic