betaflight/betaflight-tx-lua-scripts #248: having syntax error with jumper t12

nangel0, 2019-10-05:
After updating jumper t12 with lastest firmware 

from companion,
latest betaflight lua script 
and latest inav lua script

i was not able to run BF lua script for VTX or any 

other

just edited radios.lua 
added this line and and compiled .luac

local supportedRadios =
{
    ["x3"] = supportedPlatforms.x7,
    ["x7"] = supportedPlatforms.x7,
    ["x7s"] = supportedPlatforms.x7,
    ["xlite"] = supportedPlatforms.x7,
    ["xlites"] = supportedPlatforms.x7,
    ["x9lite"] = supportedPlatforms.x7,
    ["x9d"] = supportedPlatforms.x9,
    ["x9d+"] = supportedPlatforms.x9,
    ["x9e"] = supportedPlatforms.x9,
    ["x10"] = supportedPlatforms.horus,
    ["x12s"] = supportedPlatforms.horus,
    ["t12"] = supportedPlatforms.x7,
}

Just replace files in your SD card file structure 

really not shure if can do this, its the first time im adding anything to git and to community

mikeller, 2019-10-06:
@nangel0: Yes, this is exactly what was done in #231, and will be included in the next release.

raphaelcoeffic, 2019-10-06:
@mikeller just thinking loud, but while adding the virtual events, wouldn’t actually make sense to just list the screen formats as “platforms”, and rely on opentx doing the right thing? That would avoid having to add every single radio.

mikeller, 2019-10-06:
@raphaelcoeffic:

> wouldn’t actually make sense to just list the screen formats as “platforms”, and rely on opentx doing the right thing? 

If this is possible that would definitely be an advantage

raphaelcoeffic, 2019-10-06:
@mikeller I just posted a PoC in #250

nangel0, 2019-10-06:
> @nangel0: Yes, this is exactly what was done in #231, and will be included in the next release.

Sorry, new to github should have searched first...

raphaelcoeffic, 2019-10-14:
@nangel0 could you please re-test with 1.4.0? at best, directly with the zip file in #260.

raphaelcoeffic, 2019-10-14:
Support for T12 is included in release 1.4.0.

closed 2019-10-14 by raphaelcoeffic

nangel0, 2019-10-14:
Will test asap