betaflight/betaflight-tx-lua-scripts #255: X9D+ - virtual events lua still not correct

RipperDrone, 2019-10-10:
@raphaelcoeffic sry to report non-working fix for Lua - PIDs adjustment.

To replicate:
- flash latest OMNINXTF7 FW with BFC 10.6.0RC2
- load latest today's lua onto Taranis (delete all .luac files on Taranis, let them recompile automatically by Taranis when switching on)
- long press PAGE to get into telemetry page, short press to get into PID manioulation page
- edit first field, press '-' --> Field value decrements ok
- press 'ENTER' --> field value jumps back to old value

Funnily, in repeating this procedure, 2nd time trying to decrement the same value, it stays decremented.

mikeller, 2019-10-10:
Thanks @RipperDrone. I just verified again, and it's working fine on X9D+ with 2.3.0 - looks like they've broken it in the changes for 2.3.1. :frowning_face:

raphaelcoeffic, 2019-10-10:
Really strange, it’s working fine in the Simu in 2.3.1. @RipperDrone did you try with the code from the other PR?

robschii, 2019-10-11:
in als menus of the lua script    als values jump back to the value before...  i will stay at opentx 2.2.4

RipperDrone, 2019-10-11:
@raphaelcoeffic pls specify 'other PR' a little more. I have tested with y'day's .zip download of master, so both PRs should have been included in the /src directory, shouldn't they?

raphaelcoeffic, 2019-10-12:
@RipperDrone sorry, I meant #250. Please note that the ZIP file is outdated. Just use the files in `src/`.

RipperDrone, 2019-10-12:
Success! Working now that I downloaded the /src files from the correct branch. Thank you @raphaelcoeffic - and closing this issue accordingly.

Now we have to handle the 'background noise' from people NOT matching the correct versions of lua and OTX - hope there is a consistent way out :-)

closed 2019-10-12 by RipperDrone