betaflight/betaflight-tx-lua-scripts #258: lua 1.4: pages automatically incremented when swapping btw telemetry display and parameter edit pages

RipperDrone, 2019-10-13:
When swapping btw lua page #1 (telemetry display) and page #2 (parameter edit pages) on Taranis X9D+ (OTX 2.3.1, Lua 1.4.0), every time swapping back to the lua parameter edit page #2, the lua sub-page of parameter editing is incremented, e.g. editing PIDs, then swapping to tele display, swapping back to editing --> PIDs page is no longer shown, but the next edit sub-page in sequence (=PIDs 2/2: Feed Forward) etc.

Has been different behavior before: Swapping back to last edited sub-page OR always starting with 1st edit sub-page would both be intuitive. I think former behavior was restarting with 1st page every time the lua edit parameters pages are entered.

0crap, 2019-10-13:
Same here on the Taranis QX7S OpenTX 2.3.1 and Lua 1.4.0
In the past I could press the PAGE key to switch between the Lua script and User configurable Display Screens without the Lua script skipping forward.
Not a big issue but noticeably different between Lua script 1.3.0.

raphaelcoeffic, 2019-10-13:
Seems OpenTx passes the key press to the displayed page AFTER the telemetry page change has been done. Unfortunately, we cannot differentiate it from another key press here.

RipperDrone, 2019-10-13:
@raphaelcoeffic Can we not always start on parameters page 1 of the lua script? Makes it at least intuitive: When you leave the parameter edit pages, you seem to be done with changes, so when you enter them the next time, you can expect they start on page 1 again...

raphaelcoeffic, 2019-10-14:
@RipperDrone please have a look at #260. This should fix it.

RipperDrone, 2019-10-14:
@raphaelcoeffic Confirmed - this PR fixes it :-) Niiiice! :-)

closed 2019-10-14 by raphaelcoeffic