betaflight/betaflight-tx-lua-scripts #263: Lua script pages not working

Barux, 2019-10-15:
Hello,

running Taranis X9d+ SE (not 2019, old one), Betaflight TX Lua Scripts 1.4.1, OpenTX 2.3.1 (also bootloader), Betaflight 3.5.7, Crossfire micro tx + nano rx, Hobbywing G2 F4 (OmnibusF4SD)

The pages listed hereunder don't update, they keep values "---"

NOT working pages:

PIDs 2/2
Rates
PID Advanced
Filters
GPS Rescue
GPS Rescue/PIDs

Can you please check?

Thanks

mikeller, 2019-10-15:
@barux:

> Betaflight 3.5.7

You'll need to update your firmware version to make it work with the latest scripts, or downgrade the scripts to the version that was released for 3.5.

Poitivin, 2019-10-17:
Hi, I had the same exact issue on my 1st batch X9d (so very very old :) ) and after updating tx firmware/bootloader to 2.3.1, update the SD card to latest version and "installing" LUA 1.4.1, all is working OK.

Thank you very much for all those who had invested so much effort and time so others on the community also get the fun!

closed 2019-11-23 by stale[bot]