betaflight/betaflight-tx-lua-scripts #267: VTX Settings Incorrect and not editable

chadmbrown, 2019-10-19:
Running QX7S on OpenTX 2.3.1

Actual VTX settings Band F Channel 4, 5800 and power set to 800. This is what the LUA scrips shows (attached) which cannot be edited.

[image]

mikeller, 2019-10-27:
@chadmbrown: Can you please supply some more information about your setup:
- flight controller firmware version used;
- flight controller configuration (`diff`);
- VTX used;
- RX hardware used;
- RX firmware version;
- lua script version.

Nikel9O, 2019-10-31:
having the same issue here,
flight controller is a mamba f405 mini mk2 (FURYF4OSD target on BF 4,1)
lua script 1.4.1
VTX is FuriousFPV VTX Stealth Nano Race 5.8G
RX is an X-Lite running opentx 2.3.1
(note that the vtx changes channel and power perfectly on the OSD)

mikeller, 2019-10-31:
@Nikel9O: What RX are you using (on the craft)?

Nikel9O, 2019-10-31:
R-xsr through fport

mikeller, 2019-10-31:
@Nikel9O: Some older versions of the r-XSR FPort firmware have got an issue with data transmission (#151). Make sure you update your r-XSR to the latest firmware version from the FrSky website.

Nikel9O, 2019-10-31:
Already running 190311

Knockout808, 2019-11-08:
I'm having what looks like the same issue. The VTX settings can't be changed via the controller but can be change properly via OSD. My controller will display Band: U, Chan: 0, Power: 0, Freq: 0 unless you specify a band, channel, and power in the Selected Mode portion of the Video Transmitter tab in BF. However, the settings are still not editable via the controller after doing so, only through OSD. 

flight controller: Hobbywing XRotor F4 G2 OMNIBUSF4SD(AIRB) on BF 4.1.0 (I also tried OMNIBUSF4SD(LEGACY) 4.1.0 with the same result)
LUA Script 1.4.1
VTX: Mytek 5.8G VTX-HV W/ BFCMS CONTROL (Firmware V1.9)
VTX Table: I tried both the IRC Tramp USA table and the custom CLI command listed on Maytek's website with the same results.
RX: X-Lite Pro running opentx 2.3.1
Reciever: RXSR S.Port on Access FW 1.1.3

Knockout808, 2019-11-08:
I was having this same issue, I found the fix is you have to set a Band, Channel, and Power in the Selected Mode portion of the Video Transmitter tab and then the VTX settings can be changed via your controller after that. I did run into an issue though, the power levels shown in the controller dont match my VTX; 25 changes to 25mw, 100 changes to 200mw, and 200 changes to 500mw. My controller also shows 400 and 600 as selectable power levels but if I select one of those options and  save it, then it reverts back to zero and no VTX settings can be changed. I have to go back into BF and set the power level to one of the available settings (ie. 25,200, or 500 for my vtx) and then I can edit settings on my controller again. As a side note changing settings via OSD works perfectly.

flight controller: Hobbywing XRotor F4 G2 OMNIBUSF4SD(AIRB) on BF 4.1.0 (I also tried OMNIBUSF4SD(LEGACY) 4.1.0 with the same result)
LUA Script 1.4.1
VTX: Mytek 5.8G VTX-HV W/ BFCMS CONTROL (Firmware V1.9)
VTX Table: I tried both the IRC Tramp USA table and the custom CLI command listed on Maytek's website with the same results.
RX: X-Lite Pro running opentx 2.3.1
Reciever: RXSR S.Port on Access FW 1.1.3
Diff:	
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09
resource SERIAL_RX 3 NONE
resource PINIO 1 B11

# feature
feature -AIRMODE
feature SOFTSERIAL
feature TELEMETRY
feature 3D

# beeper
beeper -RX_LOST

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1775 2100 0 0
aux 1 13 1 1300 2100 0 0
aux 2 28 3 1700 2100 0 0
aux 3 29 2 1300 2100 0 0
aux 4 35 2 1825 2100 0 0
aux 5 40 3 1300 2100 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 375
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 9
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 120
set yaw_motors_reversed = ON
set 3d_deadband_throttle = 10
set pid_process_denom = 1
set osd_cap_alarm = 1500
set osd_rssi_pos = 14585
set osd_tim_2_pos = 14678
set osd_vtx_channel_pos = 14370
set osd_mah_drawn_pos = 14615
set osd_warnings_pos = 14379
set osd_avg_cell_voltage_pos = 14647
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5740
set vcd_video_system = NTSC
set pinio_box = 40,41,42,43
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225

rateprofile 3

# end the command batch
batch end

mikeller, 2019-11-09:
@Knockout808:

> Edit: I've fixed my issue, I went into the VTX.lua files and edited:
-- Tramp
self.fields[3].table = { 25, 100, 200, 400, 600 }
self.fields[3].max = 5
to
-- Tramp
self.fields[3].table = { 25, 200, 500 }
self.fields[3].max = 3
everything is working correctly for me now.

Yes, the VTX settings are not matched to the VTX table used in 1.4.1 - we are working on this (or rather working on fixing memory overrun problems caused by the fix for this), and it will be solved in the next release of the scripts.

closed 2019-12-16 by stale[bot]