betaflight/betaflight-tx-lua-scripts #271: X9 Lite support

alexdibi, 2019-10-22:
version 1.4.1 don't support x9 lite

mikeller, 2019-10-22:
@alexdibi: Can you please supply a little bit more information like what exactly is not working, or what error message you are getting?

alexdibi, 2019-10-23:
I get error: 
Script Syntax Error
BF/radios.lua 
incompatibile precompiled

mikeller, 2019-10-23:
@alexdibi: And what version of OpenTX are you running?

alexdibi, 2019-10-23:
2.3.1

mikeller, 2019-10-23:
Interesting. I don't see why this would not work. When did you download the scripts?

alexdibi, 2019-10-23:
Yesterday... i tried inserting the string ["x9lite"] = supportedPlatforms.x7
in radios.lua and the First Page work than on second Page i give an Error in functions.lua

mikeller, 2019-10-23:
@alexdibi: If you want to run your own modified version of scripts you are on your own. If you want to ask for support here try running the released version.

alexdibi, 2019-10-23:
i have tried now i'm on 1.3.0

alexdibi, 2019-10-23:
1.4.1 official don't work on x9 lite

frozenskys, 2019-10-23:
What error does the official release of 1.4.1 give on the x9 lite?

alexdibi, 2019-10-23:
@frozenskys 
Script Syntax Error
BF/radios.lua
incompatibile precompiled

frozenskys, 2019-10-23:
I don't have an x9 lite available to test on but I just downloaded the 1.4.1 release and tried it in the simulator and it works fine - no need to add anything about x9lite in radios.lua. I suggest you delete any luac files off your SD card, and delete the current radios.lua on there as well, then try installing the 1.4.1 files again.

alexdibi, 2019-10-23:
@frozenskys ok i Will try with another micro SD with only ad card contents for opentx 2.3.1 and lua 1.4.1

frozenskys, 2019-10-25:
@alexdibi did you get it fixed?

alexdibi, 2019-10-25:
@frozenskys yes there was some file put wrong in the sd ....

closed 2019-12-01 by stale[bot]