betaflight/betaflight-tx-lua-scripts #273: BF lua 1.4: long press decrement key does not make values run down, it just decrements by 1

RipperDrone, 2019-10-25:
BF 4.1. / lua 1.4 / openTX 2.3.1 / lua script running on telemetry page / Taranis X9D 2013:

- when long pressing '+' button, any edit value (PID, GPS rescue, ...) increases continuously until button is released or max of allowable range is reached.
- when pressing '-' button, it just decrements by one and then stops. Long press does not seem to work, you have to single press multiple times.

Pbly a matter of virtual events not being propagated correctly to OTX for GPS rescue page lua fields?

RipperDrone, 2019-10-25:
Hardware OmniNXT F7, CRossfire ano/Micro combo, telemetry over CRSF

frozenskys, 2019-10-25:
I think it's this in the lua scripts:

```
-- editing value
elseif currentState == pageStatus.editing then
    if event == EVT_VIRTUAL_EXIT or event == EVT_VIRTUAL_ENTER then
        currentState = pageStatus.display
    elseif event == EVT_VIRTUAL_INC or event == EVT_VIRTUAL_INC_REPT then
        incValue(1)
    elseif event == EVT_VIRTUAL_DEC or event == EVT_VIRTUAL_DEC_REPT then
        incValue(-1)
        killEvents(event)
    end
end
```

@raphaelcoeffic any idea why there is a killEvents here?

RipperDrone, 2019-10-25:
@frozenskys darn, which file is it in? I searched for EVT_VIRTUAL_DEC_RPT but could only find a symmetrical code snippet.

I think there has been problems of ''double key press' events being generated uncommanded. Pbly this is a leftover of this after it has been fixed by other means / in OTX meanwhile?

frozenskys, 2019-10-25:
Could be left from before the change to virtual events - this is in the SCRIPTS\BF\ui.lua file - I'll need to check the history when I get back from work - I'm limited to what I can do with code right now in the office... Also I checked latest master so not sure if it's in 1.4.x

raphaelcoeffic, 2019-10-25:
Yeah, looks like a left over to me as well. Or maybe on some platform it needs to not generate another key press... hard to remember what I did. Could you please try to check with “git blame” where this is coming from?

frozenskys, 2019-10-25:
@raphaelcoeffic commit 86860af6409f2477b18e314e425b49935ed0bd36

RipperDrone, 2019-10-25:
I deleted the killEvents line, but issue remains. Didn't find another occurence of EVT_VIRTUAL_DEC_REPT in the ui.lua file...

frozenskys, 2019-10-25:
Looks like this was a fix for an issue pre virtual events - see #245 and #251 
so I'd assume it's no longer needed.

RipperDrone, 2019-10-25:
PR created.

raphaelcoeffic, 2019-10-25:
@RipperDrone @frozenskys Does it work properly on X9D+ without this line?

RipperDrone, 2019-10-26:
For me it does - yep.

closed 2019-10-27 by RipperDrone