betaflight/betaflight-tx-lua-scripts #274: 'no telemetry' on Lua telemetry page

RipperDrone, 2019-10-25:
Taranis X9D+, CRSF Micro/Nano combo, telemetry over CRSF, Lua 1.4 run on telemetry page:

- values show ok
- editing/manipulating parameters and saving them back is fine

However, Taranis display constantly reads 'no telemetry' in bottom line of telemetry page. Is this due to using CRSF and it's checking for a non-CRSF telemetry transmission to NOT show this warning?

Maybe user error... :-/

frozenskys, 2019-10-25:
@RipperDrone that message is shown purely on rssi = 0 what rssi are you seeing in the telemetry?

`protocol.rssi() == 0`

RipperDrone, 2019-10-25:
@frozenskys Ah, so it IS connected to CRSF: I haven't set up RSSI which is rather meaningless as a telemetry value to monitor under CRSF, but have set up link quality as a parameter (RQly). Seems I have to add RSSI, just to make the display happy, then...?

frozenskys, 2019-10-25:
Probably in the short term - although we should really fix the scripts to work with either LQ or rssi 😄

frozenskys, 2019-10-25:
@RipperDrone actually - when digging further in to the CSRF protocol rssi is defined as:

`rssi = function() return getValue("TQly") end,`

RipperDrone, 2019-10-25:
@frozensky I verified with individual sensors on/off in CRSF: It turns out I need to set the TQly sensor to ON, then it seems to be understood as an equivalent to 'general RSSI'. Not so for RQly or any of the other link quality related CRSF sensors, like 1RSS, 2RSS, ...

Maybe some CRSF users/experts can give their opinions if this is intended functionality then or it needs a change?

frozenskys, 2019-10-25:
RQly is Uplink 
TQly is Downlink

So I'm assuming that the script is correct in checking for valid telemetry by checking the TQly...

RipperDrone, 2019-10-25:
I just read in some CRSF setup guides that RQly is the number you want to have in your OSD + telemetry to monitor range... not sure about the pro's and con's of all these LQ, 1RSS, 2RSS, R/TQly figures... :-)

frozenskys, 2019-10-25:
Me either - The way I understand it is this

- RQLy - is the RC link quality (up-link) - this is important for control :)
- TQly - is the Telemetry quality (down-link) - this is only important for GPS and vBat data etc.
- RSS - not really useful for quad pilots.

So in OSD etc you want RQly, but for checking if we have telemetry link then it's TQly.

closed 2019-11-25 by RipperDrone