betaflight/betaflight-tx-lua-scripts #282: Clean OpenTX 2.3.1 w/BF 1.4.1 = "Attempt to call upvalue"...

leonard4, 2019-11-08:
I just formatted my SD card on my X9D+, I flashed OpenTX 2.3.1 and extracted the latest SDCard contents to my sd card, and then extracted the latest BF lua files. I set the script to "bf" and restarted the radio. When I try to open the telem page I am presented with this:

[image]

I haven't had any luck with any searches on open or closed issues and google is coming up empty. Any advice or ideas?

leonard4, 2019-11-08:
I found that if I edit out x9pre.lua file and comment out the "filters.lua" page, then everything works correctly. Not sure if that helps identify the problem.

mikeller, 2019-11-08:
@leonard4: What flight controller / firmware version are you using? A CLI `diff` of the firmware would help as well.

leonard4, 2019-11-08:
Running a Bardwell F4 with BF 4.1.  I'm getting the same syntax error on two different quads, and even with no quads on, just trying to access the filter lua page itself.

klutvott123, 2019-11-09:
I'm not able to reproduce this on my QX7 or in the sim. @leonard4 could you try deleting the .luac files and see if it gives you a more detailed error message?

GarageTech, 2019-11-26:
I have the same problem, but i solved it by re-writing whole betaflight-tx-lua-scripts to SD card on PC.
Also important SAFELY remove SD card after writing. Problem in that files have wrong size during writing.

klutvott123, 2019-11-26:
@GarageTech Could you tell us more about how you were able to reproduce this issue? Might be useful information.

closed 2020-01-03 by stale[bot]