betaflight/betaflight-tx-lua-scripts #287: lua script in tools menu x12s horus

AndreasAusD, 2019-11-18:
i'm not shure if this is an issue or just a change on the documentation :

put the lua scripts on the sd-card of a x12s horus, openTx 2.3.2
Did not show up as a telemetry widget nor in the tools menu...

moved the bf.lua and bf.luac file from the /BF folder of the card to the /SCRIPTS/TOOLS directory and added a line at the beginning of the bf.lua : 
local toolName = "TNS|Betaflight Configurator|TNE" (copied from the crossfire.lua)

Now it shows up in the Tools menu and can be started from there...

klutvott123, 2019-11-18:
No issue. The tools version was added after the latest official release. If you wan't the tools version, you could try the latest master.

frozenskys, 2019-11-19:
@klutvott123 _"Did not show up as a telemetry widget"_ seems odd - as shouldn't the 1.4.x still show as a telemetry script? or is a "telemetry widget" something different?

klutvott123, 2019-11-19:
@frozenskys Not sure if there has been any updates to opentx, but the telemetry version has never worked on the horus as far as I know

closed 2019-12-26 by stale[bot]