betaflight/betaflight-tx-lua-scripts #290: Add calibration

mikeller, 2019-12-01:
With the requirement to have a valid ACC calibration if ACC is enabled introduced in betaflight/betafligh#9031, some users will be caught out, finding themselves in the field without a calibrated ACC, CMS not available (LoS pilots), and not able to figure out the required stick commands.
betaflight/betaflight#9179 addressed some of this by adding calibration to CMS. To cover all bases we should add a similar menu to the lua scripts, or add the calibration to the main menu.

@klutvott123: Any suggestions as to where to best put this? Separate page / add to a page / menu / ...

klutvott123, 2019-12-01:
Since this https://github.com/betaflight/betaflight/blob/9425831756015a81fe82b9c5a768c264c8603da1/src/main/msp/msp.c#L2543-L2548 is what's available through MSP I think the simplest solution is to do it as a menu function. Unless you're going to make some MSP changes. Would be nice to have a page like the one in your .gif where we could calibrate and see the status.

mikeller, 2019-12-01:
@klutvott123: Yes, the idea is definitely to extend MSP to at least allow calibration of gyro / ACC / baro to be performed. Not sure about the status feedback, this would require more MSP changes, and probably a new MSP command. Combined with the fact that we cannot poll MSP over SmartPort too often this will probably mean that a 'live' display like in CMS will not be feasible.

klutvott123, 2019-12-02:
In that case I think we should put it in the menu just like the reboot function. Maybe we can do a reboot to let the user know something actually happened.

RipperDrone, 2020-03-22:
Isn't calibration in the field easily done via RC commands...?: 
[image]

mikeller, 2020-03-22:
@RipperDrone: Yes it can be done, but the user needs to know the right stick combination (I carry a print out of the stick commands in my field kit). We want to make it as easy as possible to calibrate, to avoid users getting stuck in the field without being able to calibrate.

KAJE1977, 2020-05-09:
Is it possible to make a script that shows the stick combinations on your radio screen?

kafie1980, 2020-09-10:
For full size Taranis transmitters, there is an old lua script by the name Cleanflight Commands that allows running Mode 2 stick functions. Should still work with Beatflight as its the same stick combinations.

It was initially posted by Nek on RCGroups : https://www.rcgroups.com/forums/showpost.php?p=30993861&postcount=4366

The same script is also posted on: https://rcsettings.com/index.php/viewdownload/13-lua-scripts/204-cleanflight-commands

I ran it via the OpenTX Companion Simulator on an X9d model and it still works.

To get it to work, I had to comment lines 136 & 146 as the lcd.lock() function had been made redundant since OpenTX 2.1. You will also need to updated the Channel numbers for your Throttle, Yaw, Pitch and Roll axis on lines 3 to 6 based on your transmitter setup.

After editing the  Simply copy it to the \SCRIPTS\TOOLS\ folder on your sdcard and execute it via the Tools menu on your transmitter.

I only have an X-Lite S transmitter but the script does appear to run but does not display anything properly due to the differences in screen dimensions. (At least this is what my Geologist brain could assume :) ).

kafie1980, 2020-09-12:
I managed to modify the above Cleanflight Commands lua scipt (originally posted by Nek on RCGroups) and tested it to work perfectly on my X-Lite S.

Attached the lua script here as a .txt file extension but you need to change it to .lua and copy it to the \SCRIPTS\TOOLS\ folder on your sdcard and execute it via the Tools menu.

[Calibrate Model.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/5212134/Calibrate.Model.txt)

PR #375 (2020-12-19): Accelerometer calibration
Adds a special page to calibrate the accelerometer. The page can be accessed from the menu like this:
[image]
selecting "acc cal" brings up this:
[image]
where [ENTER] is the enter key and [EXIT] is the exit key on the radio. Pressing the enter key will send ```MSP_ACC_CALIBRATION``` to the flight controller. The flight controller will calibrate the accelerometer and send the command back. The page will exit automatically when the response is received.

Fixes #290
closed 2020-12-19 by mikeller