betaflight/betaflight-tx-lua-scripts #293: Unable to get VTX power labels (#70 Betaflight LUA)

skorzo, 2019-12-09:
Currently running:
Betaflight 4.1.1 (omnibus F4 V6)
OpenTx 2.3.1 (Taranis 9XD plus)
TBS Pro Race v2 (smartaudio)
Betaflight LUA compilation #70  [https://ci.betaflight.tech/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/lastSuccessfulBuild/artifact/](url)

I have set up the Vtx tables and then I put the model_name.lua in /BF/VTX

Lua Script is working fine but power labels are not showing.
I get "0" and "1" values and not my labels (25, 200, 400)

Asizon, 2019-12-09:
Same this with SA 1.0, getting 0 and 1 values. With SA 2.0 Tbs Race device I have getting 200 correct value, but cant change it to another power.

Asizon, 2019-12-09:
Sorry, but only my 2.0 device is in Betaflight 4.2 version

frozenskys, 2019-12-09:
@Asizon, @skorzo can you paste your "model_name.lua" here, and also confirm that the model_name part matches the "Model Name" in your transmitter?

skorzo, 2019-12-09:
> @Asizon, @skorzo can you paste your "model_name.lua" here, and also confirm that the model_name part matches the "Model Name" in your transmitter?

Please find it attached.

[Rooster_R9M.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/3939253/Rooster_R9M.zip)

The name matches the "Model Name",   even with the "_"  for  "Rooster_R9M"
When the name does not match, the shown power values are the legacy ones.

Thanks!

Asizon, 2019-12-09:
Model name correct, matches with transmitter model name. Also .lua file seems  show correct values
[AK200R5.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/3939594/AK200R5.zip)
.

Asizon, 2019-12-09:
With this test using SA 1.0, Lua show me  0 and 1 power values.
[AK200R4.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/3939603/AK200R4.zip)

frozenskys, 2019-12-09:
It's a bug in the vtx lua script and/or betaflight configurator. I'll look at it today and see if I can get you a fix...

klutvott123, 2019-12-09:
``powerTable = { [7]=25 , [16]=200, [25]=600, },``
The indexing is the problem. The tables are using the powervalues as indexes, but I believe these tables should be indexed from 0 as the range of ``vtx_power`` is 0 to 7. 
Also when we try getting the number of elements of a table like that using ``#`` the result will be 0. Probably because the element at index = 1 is nil.

frozenskys, 2019-12-09:
@klutvott123 yes - the index shouldn't be the index as for some SA versions it's set in dB rather than an index. Will it be best to have a `powertable` like the following?
```
powerTable = {
    { 7, "25" },
    {16, "200"},
    {25, "600"},
}
```
which would look like the following for the old SA1.0 style settings
```
powerTable = {
    {0, "25" },
    {1, "200"},
    {2  "600"},
}
```

klutvott123, 2019-12-09:
@frozenskys That would probably work. Do we need the powervalues for the lua script? Wouldn't it be easiest to just not add them to the table since we're just using the labels?

frozenskys, 2019-12-09:
@klutvott123 I need to check how the power levels work... if we are only sending the "BF power index" or label to BF then that's OK - if it's expecting the dB value or the "VTX power index" then no...

Asizon, 2019-12-09:
@frozenskys Ok, Lua's show me correct values, but only can change between 200 and 600
[image]
[image]

Asizon, 2019-12-09:
Ok, I think its typo afetr , ?

frozenskys, 2019-12-09:
@Asizon OK - but does it actually change the VTX power levels? I'm not at my bench so can't test this for myself...

You can also try 
```
powerTable = { [1]=25, [2]=200, [3]=600, },
```
and see if that gives you all three power levels?

Asizon, 2019-12-09:
Done! three power levels available, checked with Betaflight Configurator and the power changes are effective.

frozenskys, 2019-12-09:
@klutvott123 that would seem to say that we just need the labels - but need to start with an index of 1... so I'd say it's just a change to the configurator to write the correct values...

frozenskys, 2019-12-09:
Fix for the configurator looks simple - I'll give it a test when I get back and open a PR if it's OK...

klutvott123, 2019-12-09:
@frozenskys Nice! I would test if I could but I don't actually own a VTX lol.

frozenskys, 2019-12-09:
@skorzo while I'm working on the fix for the configurator, setting your `powerTable` in the lua file to the following should get it working for you...
```
powerTable = { [1]=25 , [2]=100, [3]=400, },
```

skorzo, 2019-12-09:
> @skorzo while I'm working on the fix for the configurator, setting your `powerTable` in the lua file to the following should get it working for you...
> 
> ```
> powerTable = { [1]=25 , [2]=100, [3]=400, },
> ```

Yes, it works! Label is shown and power is correctly changed. Many thanks for your job

closed 2019-12-26 by skorzo