betaflight/betaflight-tx-lua-scripts #294: Larger Text Size for 480 x 272 Displays

MRC3742, 2019-12-10:
I have just tested the current master 1.5.0 build and see that you have changed to a menu driven main page. All seems to work properly for my needs on the Jumper T16 but I see that on the larger color screens (480 x 272) that the text scaling is now much smaller than with version 1.4.1. Would returning the scaling to earlier formats on the larger screen be an issue or are you looking to add additional functions per page? It is much easier to read on the larger screen if the scaling was returned to what it has been.

I understand that currently your main concern is to add features and still reduce memory usage for the Radios with limited allocated memory available. Such as the X9D+ and similar models.

Example Text Scaling on Current Version
[image]

Example of Text Scaling on Version 1.4.1
[image]

Thanks - Rich

klutvott123, 2019-12-11:
We have moved away from setting the text size per field/label. Now everything uses the same size which is set globally through one variable. Changing it will change the size of all text and might be a problem for the parameters presented in a list format. The text size is the default which is also used in the opentx menus.

Maybe it would help if the table elements were spaced a little further apart? 
You can test this yourself by changing row/col here: https://github.com/betaflight/betaflight-tx-lua-scripts/blob/f78dba804446ffc27abafa8c49fccebb551994aa/src/SCRIPTS/BF/TEMPLATES/480x272/default_template.lua#L6
This will change the spacing for all tables.

MRC3742, 2019-12-11:
It just seems with a screen more than twice the width resolution as the X9D+ that the small font is just not appropriate. I think that it can be easily adjusted using your current page views.

I've made a few changes to the 480x272 template and the radios.lua and it seems to fill the screen more to scale. It looks better and is definitely easier to read using the MIDSIZE font.

I've attached all eleven radio page screenshots in the .bmp format zipped for review.
The screenshots seem to show some text height cuttoff problems. This is only a problem in the screenshots and all text appears correctly on the Radio screen.

[Screenshots.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/3948267/Screenshots.zip)

If this looks OK then I could do a pull request for the changes.
Thanks - Rich

[image]

[image]

MRC3742, 2020-01-16:
Closing this issue as there are no comments, either positive or negative, for this visual change.

closed 2020-01-16 by MRC3742