betaflight/betaflight-tx-lua-scripts #302: Unable to go back on pages with back function

Asizon, 2019-12-26:
I can't go back between pages, the back function is corrupted. I am using the latest master build, opentx 2.3.4, and Betaflight 4.2. The page feed button was also changed again and set to the 
`menu` key, I think it was more intuitive in the `page` key.

Asizon, 2019-12-26:
It is a #1835 discussion continue
https://github.com/betaflight/betaflight-configurator/pull/1835#issuecomment-564399611

klutvott123, 2019-12-31:
It's a result of moving to virtual events. The events are now the same for all TXs. It makes maintenance easier and reduces memory use. 
May I suggest trying the TOOLS version of the script instead? There the buttons will work just like expected. As a bonus it has the super nice main menu.

Asizon, 2020-01-01:
@klutvott123 The Tools version l looks awesome and works as expected thanks, but canI add this tools version to screen 1(acces with page key)???

klutvott123, 2020-01-01:
The tools version is made to be run as a one time script from the tools menu. The events won't work right if using it as a telemetry script.
I understand that it can be a little annoying to navigate to the tools page everytime it's needed but there's a lot of benefits to it. 
It uses less memory. 
All other scripts are halted when it's executed making it better able to coexist with other scripts.
Buttons work as expected.
When used in combination with the new rssi function script it offers the same functionality as the telemetry version while still using less memory.

Asizon, 2020-01-04:
Thanks for the clarification, closing this issue.

closed 2020-01-04 by Asizon

mikeller, 2020-01-16:
@klutvott123: Seeing this discussion, should we deprecate the telemetry version of the script, since it clearly has drawbacks? (I wasn't aware of the different memory footprint until now.)

klutvott123, 2020-01-16:
@mikeller I think we should. BF lua is a tool and belongs on the tools page IMO. It also makes more sense to just have one script that works on all radios. Should I make a PR for it?

mikeller, 2020-01-25:
@klutvott123: Yes please. We can probably add a `use_tools_menu_instead_of_this.txt` file just like we did in `BF/`.

klutvott123, 2020-01-25:
I'm on it. In this case I think it would be better to remove everything from the telemetry bf.lua and have it show the message about using tools menu instead. The .txt file makes sense for the ```BF``` folder as the script used to be launched directly from there, but no one will browse to the ```TELEMETRY``` folder. A lot of people will try using the telemetry version first as that's what everyone's used to. Displaying the message there ensures that everyone gets the message.

RipperDrone, 2020-01-26:
@klutvott123  maybe I got it wrong, but does it mean that quick tuning of PIDs, filters etc. by simply long pressing PAGE on the RC (after defining this telemetry script as screen 1 script / or as screen 2 script, then pressing MENU once more to enter screen 2) will no longer be an option? Instead, one has to navigate to the scripts/tools/... subfolder every time? I understand the point of timing + virtual events key conflicts and mem usage, however being able to PID/filter tune the quad LOS and not over the OSD (which is fiddly at times, too) was a big asset IMO... Any better workaround I might be missing?

mikeller, 2020-01-26:
@RipperDrone: I think you are missing the point here - OpenTX intended the 'telemetry' scripts to be used for what their name says: processing and display of telemetry data. What we were doing in the past was abusing this facility with a heavy-weight script that required input, both of these things that the telemetry script facility was **not** geared for. What we got out of it was being bitten by these restrictions: Constant memory overrun woes and defective key bindings.
I can see your point of the cumbersome UX of having to go through the system menu to 'Functions' to invoke the script though. I think the right way to address this will be to petition OpenTX to add a facility that allows 'hotkeys' to be assigned to functions, and then use these hotkeys to invoke the scripts.

klutvott123, 2020-01-26:
@RipperDrone The Betaflight lua script will be available in the TOOLS menu(long press menu. page number 3).

 @mikeller Easier access to the scripts on the tools menu has been implemented in opentx from 2.3.6. The TOOLS menu will now be page number 1 of the radio menu so that long menu press goes directly to tools.
long menu -> tools
long page -> telemetry

mikeller, 2020-01-27:
@klutvott123: Great! That's pretty much as simple as going to telemetry, and then paging between screens.

Asizon, 2020-01-27:
@klutvott123 this sounds good, thanks for these adjustements.

RipperDrone, 2020-01-27:
very nice - good compromise!

klutvott123, 2020-01-30:
@Asizon Would be nice to have a dedicated BF button 😄 

A script can be moved to the top of the tools menu by moving all other scripts to another folder, turn on tx and open tools menu, turn off tx, move other scripts back to tools folder. 
Then it's only two key presses to access it. No scrolling. 
Probably won't work for access radios as those have the spectrum analyzer permanently on the tools menu.

Asizon, 2020-01-31:
wow @klutvott123 This trick works very well thanks, now it's as simple as before, would there be any way to automate this order for the end user?
[image]

In the other hand, I dont know the reasson but now Lua script sows `No telemetry` message, but all Lua script works as expected.
[image]

klutvott123, 2020-01-31:
@Asizon I did some FAT research to find out exactly what's going on. Most of the steps in my previous post can be dropped. The files in a directory are numbered in the order they are created. 
So just by copying the files to the TOOLS folder one by one they will appear in that order in the tools menu. It could probably be automated with a bash script or something. But it wouldn't help for access radios as the built in tools are always put at the top of the list.

mikeller, 2020-01-31:
Maybe as a suggestion for an improvement in OpenTX the files could be ordered alpahnumerically. Then users could easily get the order by renaming the scripts to `1-Betaflight`, `2-<other script>`,...

RipperDrone, 2020-02-01:
Supported - would be very intuitive this way...