betaflight/betaflight-tx-lua-scripts #305: Can't Change PID Values ?

aburn09, 2020-01-25:
Hi Just updated my Taranis radio and bootloader X9D+ SE on Open Tx to the latest 2.3.5 version also updated the SD card and Lua scripts to 1.4.1 but now I can't change my PID values on the tx everytime I press page on the tx it shows the rssi value pressing again gives me the lua script PID settings but when I press ENT on my TX then either the plus or minus button on the TX the page returns to the rssi page again ?
Before I updated everything was working correctly but now reverting back to the backed up version still has the same results !!
This is a short youtube clip showing the problem ?

https://www.youtube.com/watch?v=mZf6NbP98JA

klutvott123, 2020-01-25:
I believe this is an issue with opentx 2.3.5. +/- now does the same as the PAGE key for some reason. It has been reported here https://github.com/opentx/opentx/issues/7309 too.

aburn09, 2020-01-25:
Thanks klutvott123 I suppose I'll have to only have one telemetey page ie just my pids until they fix the problem ?

klutvott123, 2020-01-25:
Yes. Another option is to try the latest betaflight lua master and launch the script from the tools menu. All keys work as they should in that version. If you decide to try it, I recommend removing all files from earlier versions first as there has been a lot of changes since 1.4.1

RipperDrone, 2020-01-26:
Confirming issue, OTX2.3.5 messes up +/- key functions in lua script

mikeller, 2020-01-26:
@RipperDrone:

> Confirming issue, OTX2.3.5 messes up +/- key functions in lua script

Is this when running the script as a telemetry script?

aburn09, 2020-01-26:
@RipperDrone yes if you run more than one Telemetry script page there seems to be a conflict with the plus and minus button on the taranis x9d+ ?

mikeller, 2020-01-26:
@aburn09: Running the Betaflight configuration script as a telemetry script is deprecated as it has a number of issues (key bindings not being supported in this mode among them), and support for it will be dropped in the next release (#306).

aburn09, 2020-01-26:
mikeller can I assume then that if I take my tx back one or two versions of open tx I can get two or more telemetry script pages working again as before I updated to 2.3.5 ?

mikeller, 2020-01-26:
@aburn09: You should ask this in https://github.com/opentx/opentx/issues/ - as stated in https://github.com/betaflight/betaflight-tx-lua-scripts/issues/305#issuecomment-578548435 Betaflight does not support telemetry scripts any more.

RipperDrone, 2020-01-26:
@mikeller yes, it's with a telemetry script. Using it to invoke the Taranis' gimbal as a low voltage alarm. 

If scripts are going to be dropped, how is the RC supposed to receive telemetry data, e.g. to trigger a LV alarm?

mikeller, 2020-01-26:
@RipperDrone: See https://github.com/betaflight/betaflight-tx-lua-scripts/issues/305#issuecomment-578549489.

closed 2020-01-26 by mikeller