betaflight/betaflight-tx-lua-scripts #313: NEXT/PREV keypress causes Telemetry view page changes

chronowarper, 2020-02-17:
Radio: FrSky X9D+  (SE, pre-2019)
Software/Firmware:  Clean install of OpenTX 2.3.5  (clean wipe of microSD) 
     copied BF 1.4.1 lua over

Issues:

When I'm in the BF LUA scripts page, pressing +/- would move the selected line forward/backward, but also flip the telemetry page between the BF lua and the OpenTX telemetry pages

This would just be a nuisance except... it also happens when I try to edit values, which kicks me out of edit mode.

Steps to repro:
  go to menu page 14 (display)
  set up 2 screens: 
   screen #1 - Scripts -> BF
   screen #2 - num

go to top menu, hold page to enter telemetry scripts, then press +/- 

Can WAR this by only enabling one screen and using Scripts -> BF

Additional isolation:
  This does not seem to be an issue with the if-elseif block in ui.lua in line 301~354, as I can comment out the incLine/decLine function calls on button events, and I still observe the telemetry page flips.

I'm not wholly familiar with the OpenTX/LUA, so I'm not sure where to look next. I found the telemtry menu block by looking at the diff of code checkin for #290.

klutvott123, 2020-02-17:
It's a bug in OpenTX 2.3.5. There will be a fix for it in 2.3.6

sedwo, 2020-03-23:
@chronowarper Thanks for the WAR.

@klutvott123 OpenTX 2.3.6 seems to still be quite a way off.   :/

klutvott123, 2020-03-26:
@sedwo I just tested the x9d+ in the sim using the current 2.3 branch and the +/- issue seems to be fixed

RipperDrone, 2020-03-28:
confirmed on Taranis X9D+ to work with just released OTX2.3.6 :-)

sedwo, 2020-03-28:
Confirmed on Taranis X9D+ to work with just released OpenTX v2.3.7  👍

RipperDrone, 2020-03-28:
@chronowarper think u can close this issue now :-)

chronowarper, 2020-03-28:
Awesome! closing as fixed.

closed 2020-03-28 by chronowarper