betaflight/betaflight-tx-lua-scripts #325: R2 and R3 not working with R-XSR

Sh0rtCut457, 2020-03-24:
Setup:
- BF 4.1.5
- Taranis X9-Lite with OpenTx 2.3.5
- Betaflight TX Lua Scripts 1.4.1
- R-XSR running firmware V1.1.4 - SBUS + S-port through SmartSerial

I can change the VTX channels to every band except Raceband 2 and 3. 
The script tries to retry the process but fails every time.

Is there a workaround for this issue ? I heard it happens only on R-XSRs that are not wired through  F-PORT.

mikeller, 2020-03-24:
This is a bug in the firmware of the R-XSR. Please report this bug to FrSky, there is nothing we can do to fix it.

closed 2020-03-24 by mikeller

FieserKiller, 2020-06-20:
Just an update: I ran into the exact same issue. All FrSky hardware and Software is updated to latest version:
Taranis X9D+ 2019 SE
OpenTX 2.3.9
ISRM v2.1.0
R-XSR ACCESS v2.1.0
BF4.2
Lua Script 1.5.0
-> TBS Unify Pro32 Nano 5G8 V1.1 runs fine via OSD, I can change all channels.
-> Via Lua Script RB2+3 don't work, everything else is working
So his old bug is still there :( 
I'll report that to frsky, the more people do that, the better.

FieserKiller, 2020-06-23:
@mikeller frsky support acts like they never heard of this before. Do you have a link for me where this problem is described en detail and frsky acknowledging its existence so I can send this link around if they need more details?

mikeller, 2020-06-23:
@FieserKiller: Lol. https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151

FieserKiller, 2020-06-25:
Another update: I ordered and installed two FrSky Archer M+ ACCESS into my new quads to get rid of that r-xsr issue but
bug is still there. everything works fine except switching the VTX to RB2 or 3 :/

marekwiecek, 2020-06-25:
@FieserKiller I also have tbs nano vtx, r-xsr with fport with access fw, taranis x9 lite with normally working lua scripts/telemetry. I will test this today in 1 hr and report back.

marekwiecek, 2020-06-25:
OK, tested OK. No problem with changing to R2 or R3.

Sh0rtCut457, 2020-06-25:
I never had any issue using FPort, only using SBUS. Still today i can't change to R2 or R3 using SBUS.
At some point i've just given up on this

FieserKiller, 2020-06-26:
@marekwiecek: strange. Are you running v2.1.0 on receivers and transmitter as well?

marekwiecek, 2020-06-26:
@FieserKiller Yes, both.

FieserKiller, 2020-06-26:
OpenTX 2.3.9 as well? (Sorry for that interrogation, I'm just trying to exclude possible sources of error because I'm basically out of ideas..)

marekwiecek, 2020-06-26:
Yes. Updated yesterday before flying.

klutvott123, 2020-06-26:
To see if this has anything to do with the bug discussed in #151, you guys could try setting one of your PIDs to 125 or 126 and save through the lua script. What happens?

FieserKiller, 2020-06-26:
I gave it a run:
- 124 no problem, saves this value directly
- 125 does not work. displays "retrying" after a few seconds and reloads previous value
- 126 does not work, same behavior 
- 127 no problem, saves this value directly

Sounds very much like the bytestuffing bug of 2018. The strange thing is that while I have this bug with 4 receivers (2x r-xsr and 2x Archer M+ on 2 quads), other people don't experience this bug at all with same hardware and software versions.

klutvott123, 2020-06-27:
@FieserKiller Thanks for testing. Looks like the bytestuff bug, but this time it's reversed. What happens if you set one of your pids to 125 in the configurator and try to load the lua pid page? 

If they have the same hardware with the same firmware they will most likely have this bug too. It's enough that the CRC for the MSP or data frame is 125 or 126 to make it fail. Unfortunately it looks like frsky has reimplemented this bug in the new access firmware. I'm using 2.1.0 ACCST and it's working as it should.

FieserKiller, 2020-06-27:
I can set both 125 and 126 via configurator and see both numbers in lua script displayed correctly.
I can change them to any other value (except 125+126 of course) via script as well.

klutvott123, 2020-06-27:
Great. So bytestuffed frames are handled correctly when sent from FC to RX, but not when sent from opentx to the TX module or from TX module to RX. It's the opposite from what we saw last time. Since it's an frsky firmware issue there's not much we can do about it. And who knows what hardware is affected this time. It will be up to the users to figure out as usual.

mikeller, 2020-06-28:
One thing that is new is that you can open issues for FrSky firmware on GitHub: https://github.com/FrSkyRC/Firmware-Test/issues

FieserKiller, 2020-06-28:
I opened an issue. If I can do some more analysis to help eliminating the lua script or OpenTX as possible causes of the bug feel free to ping me with instructions.