betaflight/betaflight-tx-lua-scripts #327: [GPS Rescue] descent_dist vs min_dth

0crap, 2020-04-03:
The current LUA GPS section allows you to change the "gps_rescue_descent_dist" value.
I prefer to have the possibility to change the "gps_rescue_min_dth" value in this place.
Can that be changed?

Not only me, but also a few flying buddies got confused by these two.
Which resulted in unwanted crashing of quads, during GPS Rescue testing.

klutvott123, 2020-04-09:
I'm not sure we should replace `gps_rescue_descent_dist`, but we could have `gps_rescue_min_dth` in addition to it. It would have to be added to MSP first so it might be too late for 4.2

mikeller, 2020-04-10:
@klutvott123: Yeah, that's something for 4.3.

0crap, 2020-04-13:
Too bad MSP needs an update, that's risky with 4.2 around the corner.
Hope it can be done in 4.3.

closed 2021-05-17 by mikeller