betaflight/betaflight-tx-lua-scripts #328: T16/Horus MSP_SET_RTC widget/background

pitts-mo, 2020-04-09:
I find it desirable to implement a basic widget for T16/Horus to run in the background with the primary purpose of  pushing the current time from the transmitter to the flight controller in the background similar to the behavior I see from my x9d+ with telemetry page in the background. (background.lua & data_init.lua?)

The inability to run bf.lua as telemetry page on T16/Horus is a known issue with various hurdles.

My specific target setup:
Jumper T16 Pro Hall V2 (daughter has been expressing enough interest for me to upgrade my x9d+)
Crossfire Micro TX
Crossfire Nano RX
betaflight 4.1.5+

pitts-mo, 2020-04-09:
If this is practical to implement... I see some interest in implementing some read only display of betaflight data such as VTX summary in a widget under issue #45

mikeller, 2020-04-09:
@pitts-mo: This is already addressed in the latest version, due to be released in a couple of weeks - the script has been split into a (non-telemetry) UI script, and a background script (needs to be run through a global function). If you want to test, you can get a development version of the installable scripts https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases

pitts-mo, 2020-04-09:
Sorry @mikeller , I guess I had not skimmed through the commits well enough :-)

Using nightly and setting a row in Global Fuctions within the main radio menu appears to satisfy this issue. 
Thank you very much,
-p

Example of my Global Functions Row:
GF* Tele Lua Script bfbkgd
 
Note to self: 
opentx on T16 keeps everything on the SD card :-O

closed 2020-04-09 by pitts-mo