betaflight/betaflight-tx-lua-scripts #329: Script syntax error Compile / compile.lua: 4

ApophysX, 2020-04-12:
Hi there,

after I updated the Lua to # 94
comes the message
Script syntax error
Compile / compile.lua: 4
loadscript ("Compile / scripts.lua", "bt") e

How can I fix the problem?

klutvott123, 2020-04-12:
Download it from here instead https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases.

ApophysX, 2020-04-12:
Hi there,

okay in betaflight-tx-lua-scripts_1.4.0.zip (v20200402.312)
scripts.lua is included. I had downloaded the latest build (# 94) from Jenkins. Thanks for the quick help.

closed 2020-04-12 by ApophysX

mikeller, 2020-04-13:
Not sure the Jenkins build is still working as intended. @wind0r: Can you please check this? The build should now use what is done in Azure in these lines: https://github.com/betaflight/betaflight-tx-lua-scripts/blob/0bd97f2a7263d5b82d47e6e68b9b8fe6533e1c8c/azure-pipelines.yml#L34-L40