betaflight/betaflight-tx-lua-scripts #330: betaflight lua script don't work anymore with X9D+ (2016)

rvmicide, 2020-04-17:
I have the taranis X9D+ (2016)... 

After upgraded radio to openTx 2.3.7, bf lua 1.4.1 don't work anymore.
I execute bf.lua/bf.luac directly from SDCARD/BF/ and most of the time nothing happens... 
Sometime i have error like:
Script syntax error
BF/ui.lua:359
loadScript("/SCRIPTS/BF/X9/pids1.lua", "bt" ...

i have tested with all version from 1.4.1 to 1.0.1 ... no result
i have tested with opentx 2.2.4 and bf 1.1.0 ... no result

only one configuration work, the openTX 2.2.2 from the frsky taranis site, since this firmware don't have GVar active (and all my model use it), i have downloaded and compiled openTX 2.2.2 and 2.2.1 but always without result.

Unfortunately i don't have backup of my previous configuration, so it's now impossible to run bf lua script on my taranis.

Any reason ?

klutvott123, 2020-04-17:
Could be anything. Does it work as a telemetry screen?

There has been a lot of changes to the Betaflight lua scripts since 1.4.1. Would you like to try a nightly build instead https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases?
It's installed just like before by copying the folders to the root of the sd card. Then it can be executed from the opentx TOOLS menu. I recommend removing all old betaflight lua scripts files and folders first.

rvmicide, 2020-04-19:
Thank's for your reply klutvott123

With 1.4.1
from telemetry screen, it's fail with message 
**Script syntax error
?:0
attempt to call upvalue '?' (a nil value)**

with v20200402.312
When i execute "Betaflight setup" from Tools page, 
i see something work in background then fail with message
**Script syntax error
loadScript("COMPILE/scripts.lua","bt") err
File not found**
Then I must reboot since the TX don't see the sdcard anymore.

Same result with bf.lua/bf.luac fom SDCARD/SCRIPTS/TOOLS/

With bf.lua/bf.luac fom SDCARD/SCRIPTS/TELEMETRY/  the fail message is just **"Script syntax error"**

and as expected from the telemetry page => **Use TOOLS menu instead of this**

is there a way to activate log on SDCARD for debugging purpose ? any other idea ?

Thx in advance

mikeller, 2020-04-19:
@rvmicide: Can you try if a clean reinstall fixes this - delete everything from the SD card, download and unpack the OpenTX contents of the SD card, download the Betaflight TX lua scripts and unpack them onto the freshly installed SD card.

rvmicide, 2020-04-20:
I've installed and reinstalled so many versions of the SDCARD and FIRMWARE over the last few days that I honestly didn't believe your suggestion.

But to be able to say "it's not working", I'm retaking the test.

I took a **new** SD card and I transferred the last downloaded SD card content and then the script..., test it and ......**now it works.**...  The shame is on me, what the hell is this ?

I compared the 2 SD cards but the operation fails because of a corrupted file in the SOUND directory. After a Scan/repair and a fresh transfer of the SOUND folder, both cards are now working.

I think that the quality of the priginal SDCARD has become bad, it loses data and causes errors that are difficult to connect.

It's been several weeks since I was trying to get BF script working, convinced that the problem was in the version... when it's the hardware 

Thank you very much all for your reply, sorry for this post which finally became support.

mikeller, 2020-04-20:
Yes, I find too that SD cards tend to develop errors over time when they are used - my working theory is that this is due to bad contacts, or things like being ejected while they are still in use.
Reformatting the card with the right tool (https://www.sdcard.org/downloads/formatter/) usually (not always) fixes this.

crystalmark, 2020-04-23:
After an upgrade to 2.3.7, I found the hard way that the default USB connection action was Storage instead of Joystick.  I unmounted the Taranis and SD card before pulling the usb out but it always remounted so killed my SD card.  Tried reformatting but it was well and truly dead.  Now I know to check the Radio Settings for USB Joystick before plugging in the USB lead.

mikeller, 2020-04-25:
@crystalmark: Try re-formatting the 'killed' SD card with sdcard.org/downloads/formatter - this sometimes brings them back.

realhuno, 2020-05-16:
Need help. what can i do?? 
Chorus Tracker LUA script not working... 
https://github.com/Smeat/Chorus32-ESP32LapTimer/blob/dev/ChorusTelemetry/lua/chorus.lua 

[image]

need this for our chorus32 tracker and the laptime in the osd.
https://github.com/Smeat/Chorus32-ESP32LapTimer
[image]
[image]

mikeller, 2020-05-16:
@realhuno: You are in the wrong place here to ask for help for this script - this is the issue tracker for the Betaflight configuration lua script - maybe you can find help for your script if you search on RCGroups?

realhuno, 2020-05-16:
@mikeller why did it work before and not now? a little bit of help would be nice.

mikeller, 2020-05-16:
@realhuno: As I said before, the script that you are using is not maintained by the people in here - you will have to go and find whoever maintains it and ask them.

FordsDrone, 2020-06-17:
Hello I hope I’m not hijacking this thread but seems my issue may be similar. I’m trying to run the BF Lua script v. 1.5.0 with my Jumper T16 running Opentx 2.3.9. I have done the following. Formatted the SD card using the Sdcard.org format tool. Then got a fresh copy of the SD card contents. I believe the latest is 2.3v0029. Then added the BF 1.5.0 Lua script to the SDcard root files. Then I run the BF script from the tools menu and at first it did some compiling but then when I try to run it again it’s stuck in “initializing”  I am getting telemetry from the receiver and also am able to change my vtx through the goggles. Any help would be appreciated. 
[image]

klutvott123, 2020-06-17:
@FordsDrone Connect power to your quad. It needs a connection. If that doesn't work, try updating the RX firmware.

FordsDrone, 2020-06-17:
Ok I had power to it but updating the RX May be the problem. I have the internal multi module in the jumper t16 pro but I’m not able to update it. Tried different firmwares and no good. I’ll research that issue somewhere and get back here. Thanks.

klutvott123, 2020-06-17:
Which receiver is it? What's the firmware version on it?

FordsDrone, 2020-06-21:
Solved!! Just in case someone with a Jumper T16 pro runs into the same issue I had. The fix is to update the Multi Protocol module firmware. Mine had the old version without a bootloader so I followed a Joshua Barbwell video on how to update it. Once the update was complete. Betaflight Lua scripts works. Thanks for the help.

gc9n, 2020-07-07:
same issue ,   "Use TOOLS menu instead of this"
when i enter BF.lua

klutvott123, 2020-07-07:
@gc9n Select "Betaflight setup" from the tools menu.

gc9n, 2020-07-07:
[image]

gc9n, 2020-07-07:
> @gc9n Select "Betaflight setup" from the tools menu.

I think i dont have Tools Menu!

gc9n, 2020-07-07:
forget about i found it ! , i was on wrong firmware
Thanks a lot

closed 2020-08-16 by stale[bot]