betaflight/betaflight-tx-lua-scripts #337: 4.2 not working

sparr88, 2020-05-06:
Jumper t16 with TBS nightly build  of opentx used latest Lua script without issue on 4.1 ..  on 4.2 rc1 I get no response from Lua script and am unable to operate Lua at all

joevo2, 2020-05-06:
bf 4.2 will need the new bf lua script v1.5.0 but not sure when is releasing

KAJE1977, 2020-05-09:
Have you tried the nightly build of lua scripts?

sparr88, 2020-05-09:
No please share !!

KAJE1977, 2020-05-09:
its on the betaflight main page or try jenkins betaflight

KAJE1977, 2020-05-09:
https://github.com/betaflight

KAJE1977, 2020-05-09:
https://ci.betaflight.tech/view/all/

KAJE1977, 2020-05-09:
with the new lua scripts they use a tool menu instead of the script page.

KAJE1977, 2020-05-09:
Contributor
klutvott123 commented 22 days ago
Could be anything. Does it work as a telemetry screen?

There has been a lot of changes to the Betaflight lua scripts since 1.4.1. Would you like to try a nightly build instead https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases?
It's installed just like before by copying the folders to the root of the sd card. Then it can be executed from the opentx TOOLS menu. I recommend removing all old betaflight lua scripts files and folders first.

I grabbed that from the issues on here. Others are having problems also. I think the nightly build will help. alot has changed and is much nicer now.

KAJE1977, 2020-05-09:
Just make sure you remove all the old files first so there is no conflicts.

joevo2, 2020-05-19:
4.2 support :) great job https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/1.5.0

closed 2020-06-28 by stale[bot]